ControlSystemStudio/phoebus #436: Phoebus applications behaviour

shroffk, 2018-11-19:
Observations from the epics workshops and demos
These are all from a mac machine with the java 9.0.1...I will be redoing all these tests on my windows machine once I am back. This is to note down the observations which can latter be converted to issues once they are confirmed.

Display builder
 - on mac delete on an widget does not remove it
 - no "delete" in the context menu on a widget in display builder (you have to remove widget from the widget tree)
 
Databrowser
 - changes to title, background, legend, and other misc properties are not applied to the open databrowser plot.
 - changes to the Misc do not mark the plt as dirty
 - changes to the Misc are not saved even with the save properties

General
 - autocomplete sometime strangely reset the typed pv

kasemir, 2018-11-26:
Regarding the databrowser 'misc' tab: Cannot reproduce. Changing one of the fonts or the show-legend option triggers the 'dirty' state. Maybe you had the "Save Changes" option disabled?

kasemir, 2018-11-26:
Context menu entries to copy, paste, delete widgets have been added

kasemir, 2018-11-26:
Sorry, was about to create the pull request when I noticed I had been on the master branch.
Anyway, the 'delete' and context menu has been added.
The Data Browser issue might just be about having de-selected Misc ->  Save Changes.

The autocomplete is iffy. Need an exact description of what the user is typing or clicking.

shroffk, 2018-11-26:
The databrowser functions fine on my windows machine (none of the mentioned issues are present). I will double check the mac version - I do remember ensuring that the save changes was enabled.

The context menu for the widgets in the editor could benefit with having the 
copy, delete, move back, move front commands.

Sorry for the plethora of comments...I am hoping this makes the deployment smoother if I can try to use phoebes like one of the end user would.

kasemir, 2018-11-26:
Cannot reproduce the data browser 'misc' issue on Mac, either.

The context menu now has copy, delete, paste, move back/front.

kasemir, 2018-11-26:
General comment on the context menu: Its content changes depending on what's selected, usually resulting in about 10 entries. Front/back were already in the toolbar, plus the toolbar has alignment options, undo/redo, zoom. Adding them all to the context menu would add another 15 items, which makes it less useful.

On the autocomplete, I'm sure there are remaining issues, but need a specific example.

shroffk, 2018-12-04:
More observations:

With X forwarding on Debian machines the menu dropdown does not appear in alignment with mouse click.

[image]

kasemir, 2018-12-04:
Should that be a separate issue?
Request for context menu in display builder was an actionable item which I could handle.
Menu behaviour with a specific Linux setup is very different, more like #256 or #353 or #367 .
Still, any difference when using `-Djdk.gtk.version=2` vs `-Djdk.gtk.version=3`?
I'm using GTK2 because otherwise Drag & drop doesn't work, i.e. hard to arrange tabs or edit displays.

kasemir, 2018-12-11:
>  autocomplete sometime strangely reset the typed pv

Autocomplete UI had several issues like #469, which have been fixed.

What's left on this issue is the misplaced menu dropdown seen on at least Debian.

kasemir, 2019-02-01:
Display Editor:
"Delete" key pressed while editing a property can still be problematic. In case the property text field looses focus, the widget is deleted. Not fatal because "Undo" will restore the widget, but best solution for now might be to simply ignore "Delete" for widget deletion. So require context menu "Delete" option or the Ctrl-X key combination to delete widgets. Plain delete key as often used when editing text will not delete widgets.

TimGuiteDiamond, 2019-11-15:
Hi, I just installed the latest copy of Phoebus to try it out and have to say that this absolutely baffled me. The **DELETE** key deletes things in every single editor I have ever used when a widget or selection is highlighted, why change that? **CTRL+X** is a cut in my mind, which can also be used on text so I don't see it being any less likely to be mistakenly used on a widget when aiming for text or vice versa.

kasemir, 2019-11-15:
Flabbergasted that you're baffled, so for now DELETE is again supported.

TimGuiteDiamond, 2019-11-15:
Thanks Kay, perhaps we are just used to different things but this works across many tools and even in CS-Studio so just seems a strange thing to deprecate at this point.

kasemir, 2019-11-15:
I'm aware of the widespread use of DELETE, and as noted above, it was supported in the display editor, but then disabled because people who actually use CS-Studio/phoebus in production complained that they more often than not accidentally deleted widgets. Ask two different users, and you get three different pieces of guidance...

georgweiss, 2019-11-15:
I strongly support making the DELETE key behave according to its function in most other contexts and applications.

shroffk, 2021-03-09:
The only open problem (X11 and menu) has its own Issue

closed 2021-03-09 by shroffk