ControlSystemStudio/phoebus #437: Viewer3dParserTest Failure

kingspride, 2018-11-20:
Hello,

I'm trying to deploy Phoebus on Windows Machines.
I'm building with maven 3.6.0 and openJDK 11.0.1 on Windows.

```
mvn clean verify -f dependencies\pom.xml -Dajvafx.platform=win
mvn clean install -Djavafx.platform=win
```

Currently I'm only able to build successfully when disabling tests with -DskipTests, otherwise it would fail when running the 3dViewerTests. 

```maven
[INFO] ----------------------< org.phoebus:app-3dViewer >----------------------
[INFO] Building app-3dViewer 0.0.1-SNAPSHOT                              [1/24]
[INFO] --------------------------------[ jar ]---------------------------------
[INFO]
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ app-3dViewer ---
[INFO] Deleting C:\Users\William\Desktop\phoebus-master\app\3dViewer\target
[INFO]
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ app-3dViewer ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 10 resources
[INFO] Copying 0 resource
[INFO]
[INFO] --- maven-compiler-plugin:3.6.2:compile (default-compile) @ app-3dViewer ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 8 source files to C:\Users\William\Desktop\phoebus-master\app\3dViewer\target\classes
[INFO]
[INFO] --- maven-resources-plugin:2.6:testResources (default-testResources) @ app-3dViewer ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] skip non existing resourceDirectory C:\Users\William\Desktop\phoebus-master\app\3dViewer\src\test\resources
[INFO] Copying 0 resource
[INFO]
[INFO] --- maven-compiler-plugin:3.6.2:testCompile (default-testCompile) @ app-3dViewer ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 2 source files to C:\Users\William\Desktop\phoebus-master\app\3dViewer\target\test-classes
[INFO]
[INFO] --- maven-surefire-plugin:2.20:test (default-test) @ app-3dViewer ---
[INFO]
[INFO] -------------------------------------------------------
[INFO]  T E S T S
[INFO] -------------------------------------------------------
[INFO] Running org.phoebus.app.viewer3d.Viewer3dParserTest
[ERROR] Tests run: 6, Failures: 0, Errors: 2, Skipped: 0, Time elapsed: 0.604 s <<< FAILURE! - in org.phoebus.app.viewer3d.Viewer3dParserTest
[ERROR] buildStructure_goodInputWithComment_returnNewStructure(org.phoebus.app.viewer3d.Viewer3dParserTest)  Time elapsed: 0.145 s  <<< ERROR!
java.util.InputMismatchException
        at org.phoebus.app.viewer3d.Viewer3dParserTest.buildStructure_goodInputWithComment_returnNewStructure(Viewer3dParserTest.java:46)

[ERROR] buildStructure_goodInput_returnNewStructure(org.phoebus.app.viewer3d.Viewer3dParserTest)  Time elapsed: 0.004 s  <<< ERROR!
java.util.InputMismatchException
        at org.phoebus.app.viewer3d.Viewer3dParserTest.buildStructure_goodInput_returnNewStructure(Viewer3dParserTest.java:74)

[INFO]
[INFO] Results:
[INFO]
[ERROR] Errors:
[ERROR]   Viewer3dParserTest.buildStructure_goodInputWithComment_returnNewStructure:46 » InputMismatch
[ERROR]   Viewer3dParserTest.buildStructure_goodInput_returnNewStructure:74 » InputMismatch
[INFO]
[ERROR] Tests run: 6, Failures: 0, Errors: 2, Skipped: 0
```

kingspride, 2018-11-20:
thanks for the quick answer. 

I tested the linux build aswell (under WSL) and it was fine. Mac crosscompile and windows crosscompile produce the same error as showed.

I'm using the german locale so pi is 3,14 contrary to the US 3.14 ...

kasemir, 2018-11-20:
That's it.

In the Viewer3d source code, find where the `Scanner` is created, like here:
https://github.com/shroffk/phoebus/blob/master/app/3dViewer/src/main/java/org/phoebus/applications/viewer3d/Viewer3d.java#L242
Then try adding this:
```
scanner.useLocale(Locale.ROOT);
```

When reading the *.shp file, we need to use a fixed number format. After all, in the source code we also use a fixed number format like "3.12e12" for exponential numbers, and not a localized version like "3.12 x 10^12".

kingspride, 2018-11-22:
hey, I had to `import java.util.Locale` aswell but it indeed fixed it.

sadly, I now get another error :grin:

```
[...]
2018-11-22 17:53:44 CONFIG [org.phoebus.pv] PV 'loc://x' closed
[INFO] Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 18.998 s - in org.phoebus.applications.alarm.server.FilterUnitTest
[INFO]
[INFO] Results:
[INFO]
[ERROR] Failures:
[ERROR]   AutomatedActionTest.testAutomatedEmailFollowup:228->checkActionDelay:91
[INFO]
[ERROR] Tests run: 31, Failures: 1, Errors: 0, Skipped: 2
[INFO]
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for app-3dViewer 0.0.1-SNAPSHOT:
[INFO]
[INFO] app-3dViewer ....................................... SUCCESS [  6.121 s]
[INFO] app-display-representation-javafx .................. SUCCESS [  9.529 s]
[INFO] app-display-runtime ................................ SUCCESS [ 13.444 s]
[INFO] app-display-editor ................................. SUCCESS [  5.311 s]
[INFO] app-alarm .......................................... SUCCESS [  0.093 s]
[INFO] app-alarm-model .................................... SUCCESS [ 20.356 s]
[INFO] app-alarm-ui ....................................... SUCCESS [  4.874 s]
[INFO] app-alarm-logging-ui ............................... SUCCESS [  1.740 s]
[INFO] app-scan ........................................... SUCCESS [  0.101 s]
[INFO] app-scan-model ..................................... SUCCESS [ 10.248 s]
[INFO] app-scan-client .................................... SUCCESS [  1.005 s]
[INFO] app-scan-ui ........................................ SUCCESS [  2.301 s]
[INFO] app-channel ........................................ SUCCESS [  0.093 s]
[INFO] app-channel-channelfinder .......................... SUCCESS [  1.390 s]
[INFO] app-channel-utility ................................ SUCCESS [  0.421 s]
[INFO] app-channel-views .................................. SUCCESS [  0.973 s]
[INFO] app-perfmon ........................................ SUCCESS [  0.585 s]
[INFO] product ............................................ SUCCESS [ 13.758 s]
[INFO] services ........................................... SUCCESS [  0.094 s]
[INFO] service-alarm-server ............................... FAILURE [01:10 min]
[INFO] service-archive-engine ............................. SKIPPED
[INFO] service-scan-server ................................ SKIPPED
[INFO] service-alarm-logger ............................... SKIPPED
[INFO] service-alarm-config-logger ........................ SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  02:43 min
[INFO] Finished at: 2018-11-22T17:53:45+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.20:test (default-test) on project service-alarm-server: There are test failures.
[ERROR]
[ERROR] Please refer to C:\Users\William\Desktop\phoebus-master\services\alarm-server\target\surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date]-jvmRun[N].dump, [date].dumpstream and [date]-jvmRun[N].dumpstream.
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :service-alarm-server
```

kasemir, 2018-11-24:
The AutomatedActionTest.testAutomatedEmailFollowup error in checkActionDelay should be intermittent. It's unfortunately a test that depends on timing, and sometimes the test fails because the code executes slowly. The timing constraints were recently weakened (93e50c8575bece65e3b852dc83a83f5db0b3bb69) but it looks like that's not sufficient. In any case, that's different from the locale specific issue in the 3D file parser.

kingspride, 2018-11-26:
sorry to bother you again with this, but I still have problems / questions.

I now try cross-compiling a windows build on an english linux machine and the 3dViewer Test fails with the following:

my goal is to compile versions for all 3 platforms, with jenkins. I get working versions when I skip the tests.

```
[INFO] ----------------------< org.phoebus:app-3dViewer >----------------------
[INFO] Building app-3dViewer 0.0.1-SNAPSHOT                             [35/58]
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ app-3dViewer ---
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ app-3dViewer ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 11 resources
[INFO] Copying 0 resource
[INFO] 
[INFO] --- maven-compiler-plugin:3.6.2:compile (default-compile) @ app-3dViewer ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 9 source files to /root/phoebus/app/3dViewer/target/classes
[INFO] 
[INFO] --- maven-resources-plugin:2.6:testResources (default-testResources) @ app-3dViewer ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] skip non existing resourceDirectory /root/phoebus/app/3dViewer/src/test/resources
[INFO] Copying 0 resource
[INFO] 
[INFO] --- maven-compiler-plugin:3.6.2:testCompile (default-testCompile) @ app-3dViewer ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 2 source files to /root/phoebus/app/3dViewer/target/test-classes
[INFO] 
[INFO] --- maven-surefire-plugin:2.20:test (default-test) @ app-3dViewer ---
[INFO] 
[INFO] -------------------------------------------------------
[INFO]  T E S T S
[INFO] -------------------------------------------------------
[INFO] Running org.phoebus.app.viewer3d.Viewer3dParserTest
Graphics Device initialization failed for :  es2, sw
Error initializing QuantumRenderer: no suitable pipeline found
java.lang.RuntimeException: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at com.sun.javafx.tk.quantum.QuantumRenderer.getInstance(QuantumRenderer.java:280)
	at com.sun.javafx.tk.quantum.QuantumToolkit.init(QuantumToolkit.java:222)
	at com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:260)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at javafx.application.Platform.startup(Platform.java:101)
	at org.phoebus.app.viewer3d.Viewer3dParserTest.<clinit>(Viewer3dParserTest.java:35)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
	at org.junit.runners.BlockJUnit4ClassRunner.createTest(BlockJUnit4ClassRunner.java:217)
	at org.junit.runners.BlockJUnit4ClassRunner$1.runReflectiveCall(BlockJUnit4ClassRunner.java:266)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.BlockJUnit4ClassRunner.methodBlock(BlockJUnit4ClassRunner.java:263)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.apache.maven.surefire.junit4.JUnit4Provider.execute(JUnit4Provider.java:365)
	at org.apache.maven.surefire.junit4.JUnit4Provider.executeWithRerun(JUnit4Provider.java:272)
	at org.apache.maven.surefire.junit4.JUnit4Provider.executeTestSet(JUnit4Provider.java:236)
	at org.apache.maven.surefire.junit4.JUnit4Provider.invoke(JUnit4Provider.java:159)
	at org.apache.maven.surefire.booter.ForkedBooter.invokeProviderInSameClassLoader(ForkedBooter.java:386)
	at org.apache.maven.surefire.booter.ForkedBooter.runSuitesInProcess(ForkedBooter.java:323)
	at org.apache.maven.surefire.booter.ForkedBooter.main(ForkedBooter.java:143)
Caused by: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
[... 95 lines cut ...]
[INFO] service-alarm-logger ............................... SKIPPED
[INFO] service-alarm-config-logger ........................ SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  02:37 min
[INFO] Finished at: 2018-11-26T11:06:00+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.20:test (default-test) on project app-3dViewer: There are test failures.
[ERROR] 
[ERROR] Please refer to /root/phoebus/app/3dViewer/target/surefire-reports for the individual test results.
[... cut ...]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :app-3dViewer
```

is this normal behaviour because a windows javafx cant be tested under linux ?

kasemir, 2018-11-26:
The parser is currently implemented within the 3D Viewer. So running the parser means creating the actual viewer, which depends on JavaFX graphics. The latter are platform dependent.

So on windows, you need the windows binaries for JavaFX to run the test. Means you can't execute the test when cross-compiling, because then you have the JavaFX binaries for the target OS, which won't run on the host OS.

I'll pull the parser out of the UI code.

kasemir, 2018-11-26:
Scratch "I'll pull the parser out of the UI code". The parser creates UI elements, so it can't run without the UI. I've turned the test into a demo.

Created separate issue for the occasional AutomatedActionTest failure, #439