ControlSystemStudio/phoebus #461: Remove UI dependency from core-framework

kasemir, 2018-12-05:
There's core-framework and core-ui.

For the most part core-framework has non-UI utilities (preferences, persistence, jobs, macros, NLS, autocomplete lookup, ..) and core-ui has UI deopendent utilities (common dialogs, autocomplete UI, ..).
Except that core-framework contains MenuEntry and ToolbarEntry, which in turn have `Image getIcon`. The AppDescriptor also has a `getIcon` method, but it returns a URL and is thus not UI dependent.

As a result of the core-framework UI depedence, the scan server, which uses core-framework for jobs, preferences etc., includes javafx jars. In the ant build they are simply skipped and it "works" without them, but the maven build includes them, and overall a clean separation would be better.

Option 1:
Keep MenuEntry, ToolbarEntry in core-framework, but have them return an icon URL instead of Image.

Option 2:
Move MenuEntry, ToolbarEntry from core-framework to core-ui.

shroffk, 2018-12-05:
I think option 2 makes more sense since menu and toolbars are ui elements themselves. I can't think of a situation where menu and toolbar would be used in a non-ui setting.

shroffk, 2018-12-05:
I can move these

kasemir, 2018-12-05:
OK on moving Menu & ToolbarEntry.

Do we leave the AppDescriptor, AppResourceDescriptor in the core-framework?
For now we only offer the UI PhoebusApplication, but we could have a non-UI application with shell that looks for an AppDescriptor to run, so you get a command line application that starts/stops, loads/writes memento, opens resources.

shroffk, 2018-12-05:
I agree with your analysis on the AppDescriptor & AppResourceDescriptor 
We should leave those in the core-framework

closed 2018-12-07 by kasemir