ControlSystemStudio/phoebus #468: Simultaneous logging fails for alarm Command & State messages

tanviash, 2018-12-06:
2018-12-06 14:27:34.156  WARN 351 --- [-StreamThread-2] o.a.k.c.c.internals.ConsumerCoordinator  : [Consumer clientId=streams-Accelerator-alarm-cmd-230d31ff-27d9-4f56-9aaa-2d5d46939030-StreamThread-2-consumer, groupId=streams-Accelerator-alarm-cmd] The following subscribed topics are not assigned to any members: [AcceleratorState] 

When simultaneous logging is required by alarm logger for state and command messages, only one of the 2 gets logged. See the alarm logger warning above.

kasemir, 2018-12-06:
That must be a side effect of how you're creating the consumer, maybe some detail in the streams implementation.

Works fine when creating a plain `KafkaConsumer` that subscribes to a list of topics.
Here's the code for the UI client, connecting to two topics (https://github.com/shroffk/phoebus/blob/master/app/alarm/model/src/main/java/org/phoebus/applications/alarm/client/AlarmClient.java#L73):
```
final List<String> topics = List.of(config_topic,
                                    config_name + AlarmSystem.STATE_TOPIC_SUFFIX);
consumer = KafkaHelper.connectConsumer(server, topics, topics);
```

Here's the KafkaHelper.connectConsumer detail:
https://github.com/shroffk/phoebus/blob/master/app/alarm/model/src/main/java/org/phoebus/applications/alarm/client/KafkaHelper.java#L56, which basically does this:
```
final Consumer<String, String> consumer =
   new KafkaConsumer<>(props, deserializer, deserializer);
...
consumer.subscribe(topics, crl);
```

shroffk, 2018-12-06:
I have found the issue.
The properties map given to the builder needs to be copied before it is passed to it.
Right now, since I use the same properties object the application.id gets duplicated...I will fix this and push the patch

tanviash, 2018-12-07:
Still not working with the new fix. Complains that alarm logger can't connect to kafka broker. Kafka brokers are up. Alarm-server and config-logger are connected to the same kafka nodes.

Following warnings in the alarm logger.
2018-12-07 13:31:35.129  WARN 863 --- [428905172-admin] org.apache.kafka.clients.NetworkClient   : [AdminClient clientId=streams-Accelerator-alarm-state-ecde0558-d05a-451d-9324-351428905172-admin] Connection to node -1 could not be established. Broker may not be available.
2018-12-07 13:31:35.540  WARN 863 --- [e9459e05d-admin] org.apache.kafka.clients.NetworkClient   : [AdminClient clientId=streams-Accelerator-alarm-cmd-40db7bca-c53e-42f4-89c9-ee4e9459e05d-admin] Connection to node -1 could not be established. Broker may not be available.

Topics on kafka nodes are the following.
deployer@kafkatest01:~$ kafka-topics --zookeeper localhost:2181 --list | grep Accelerator
Accelerator
AcceleratorCommand
AcceleratorState
AcceleratorTalk

shroffk, 2018-12-07:
Are you sure the properties/preferences for the bootstrap server are correct?
I did not change any code regarding the kafka connections

shroffk, 2018-12-07:
PR #475 should address the above problem

tanviash, 2018-12-07:
@shroffk Works great now. Thanks !!

closed 2018-12-07 by tanviash