ControlSystemStudio/phoebus #489: Issue with connecting PVs

tanviash, 2018-12-12:
Hi Kay & Kunal,

We are trying to connect real pvs to phoebus alarm tree and see that it disconnects after few seconds. However, the 'probe' feature of this PV does show the current value of this PV. Our settings.ini file contains the following line.
org.phoebus.pv.ca/addr_list=controlgw1.nscl.msu.edu

Also, see the following picture.
[image]

Alarm server logs show the following.
/Accelerator > 2018-12-11 13:48:31 WARNING [org.phoebus.applications.alarm] /Accelerator/Test/REA_OPS:TEST:ENUM_ALRM connection timed out
2018-12-11 13:51:21 WARNING [org.phoebus.applications.alarm] /Accelerator/Test/REA_OPS:TEST:ENUM_ALRM connection timed out

shroffk, 2018-12-12:
Hi Tanvi, 
The phoebus client and the alarm sever are independent entities...but they use the same connection library.

When you are running the alarm server, are you using the same settings.ini file too?

tanviash, 2018-12-12:
Got it working now. Issue was alarm server didn't have ca/addr_list set. It was set only on client side. Also, I needed to add the port number along with IP.

However, one thing I noticed, even after I restarted my alarm server with updated settings.ini file, PV on the client side still showed as disconnected. I had to delete the PV and re-add to the alarm tree to get it working.

Thank you both.

closed 2018-12-12 by tanviash

kasemir, 2018-12-12:
> still showed as disconnected
Are you sure?

It should show that there was an alarm, but that the current state of the PV is connected.
That's the whole point of the alarm system, to remember that there was an alarm.
If you acknowledge the alarm, AND the PV is good, then the alarm would clear.