ControlSystemStudio/phoebus #495: Display Builder widget category "user"

pavel-ch, 2018-12-13:
In the Display Builder Editor, I would rather group my site specific widgets under a user specific category. I understand that it is some work to make categories extensible, so I instead suggest simply add a category "user" to the  WidgetCategory class in basic product like this:

```
....
    STRUCTURE(Messages.WidgetCategory_Structure),
    // More categories?
    /** Widgets that do not fit the other categories */
    MISC(Messages.WidgetCategory_Miscellaneous);
    /** Widgets grouped as a site specific collection*/
    USER(Messages.WidgetCategory_User);

    final private String description;
....

```
I can do it but need a go/not go.

claudio-rosati, 2018-12-13:
I have the same problem.
I suggest "Extensions" rather than "User", because that is what they are. Moreover "User" is always associated to the current user of the application, while those are site specific extensions.

I also would like to have hangouts about site specific widgets, in order to understand if some of them are so interesting to be promoted into the community edition.

pavel-ch, 2018-12-13:
Yes I agree Extensions suits better. Further tuning can be done in messages.properties anyway.

pavel-ch, 2018-12-13:
Thinking further, I supposed this category being the last one. But, maybe, site specific widgets can be important for the site, so the question is if this category should be the first one?

claudio-rosati, 2018-12-13:
The problem is that this category will probably contain a mixture of monitors, controllers, etc.
My current single extension (an externally controllable knob) is currently into the Controls category.
Anyway, since I've added the search capability in the palette, I almost never use anymore the categories themselves, I always search typing few characters. So wherever this new category will be doesn't really matter.

pavel-ch, 2018-12-13:
Ok I vote for Extensions being the first category.
Hide category if not used like this in Palette.java (not looking to efficiency being executed once):
```
    private Map<WidgetCategory, Pane> createWidgetCategoryPanes(final Pane parent)
    {
        final Map<WidgetCategory, Pane> palette_groups = new HashMap<>();
        final Set<String> deprecated = Preferences.hidden_widget_types;
        for (final WidgetCategory category : WidgetCategory.values())
        {
            if (!WidgetFactory.getInstance()
            .getWidgetDescriptions()
            .stream()
            .filter(desc -> !deprecated.contains(desc.getType()))
            .filter(desc -> desc.getCategory() == category)
            .findFirst().isPresent())
                continue;
            final TilePane palette_group = new TilePane();
            palette_group.getStyleClass().add("palette_group");
...
```
This way, the clean installations do not even see that the Extensions category exists.

kasemir, 2018-12-13:
For the most part I don't see why custom widgets can't just use the normal categories.
Do you have a concrete example for a site-specific widget that's not a plain GRAPHIC,
or a read-only MONITOR, or an interactive CONTROL?

Still, I'm OK with adding a new EXTENSIONS category, and suppressing empty categories is certainly useful.

Not sure about putting EXTENSIONS first.
So how about placing EXTENSIONS between STRUCTURE and MISC?

claudio-rosati, 2018-12-13:
For me is fine.

pavel-ch, 2018-12-13:
Ok agreed.
To the reasons:
For example, my widgets contain a model of an electrical power switch, monitoring double-bit state, some status (qos) bits and having an open command, several variants of close command in a menu (with synchrocheck, direct etc.). It is both a monitor and control. May be misc but this tells nothing to the user.
There is a second reason: my users are encouraged to use specialized site widgets preferably
And may be third: constructing a display, it can be good to see which widgets will work in a general Phoebus installation and which not (are site specific)

closed 2018-12-14 by kasemir