ControlSystemStudio/phoebus #501: ClosedFileSystemException build error with latest Eclipse 2018-09

kasemir, 2018-12-15:
Downloaded the current Eclipse IDE (2018-09). With openjdk-11, each build fails with a `ClosedFileSystemException`.

Found hints that this is a known bug, need to install "Java 11" support (Example:
https://stackoverflow.com/questions/52540355/eclipse-2018-09-goes-into-endless-build-loop-with-java-11-plugin-closedfilesyst )

Tried to install the Java 11 support in a few ways, but it fails, for example like then when using Help/Marketplace:

[image]

berryma4, 2018-12-15:
2018-12 release will support java 11. Tycho 1.3, which supports java 11, was released yesterday.

kasemir, 2018-12-15:
No such problem with the previous Eclipse Photon 4.8.

To build with java 11, there is the issue of JavaFX now being one of the dependencies.
Since it's OS-specific, on Windows that means adding this to `dependencies/phoebus-target/.classpath`:
```
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-base-11-win.jar"/>
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-base-11.jar"/>
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-controls-11-win.jar"/>
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-controls-11.jar"/>
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-fxml-11-win.jar"/>
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-graphics-11-win.jar"/>
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-graphics-11.jar"/>
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-media-11-win.jar"/>
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-media-11.jar"/>
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-swing-11-win.jar"/>
       <classpathentry exported="true" kind="lib" path="target/lib/javafx-web-11-win.jar"/>
```

On Linux or Mac it's similar but replacing "-win" with the respective OS.

kasemir, 2018-12-15:
Also, some code that uses JavaFX didn't explicitly add JFX to the classpath. Fixed by exporting `phoebus-target` (which includes JFX) from `core-ui`, requiring only phoebus-target/.classpath to hold the OS-specifig JFX jar references.

kasemir, 2018-12-15:
Finally, when using Java 11, it seems necessary to not only configure JDK 11 as the JDK in the Eclipse Java/JRE settings, but also to run Eclipse itself with JDK 11.
Otherwise it will all compile fine, but when launched the runtime won't initialize JFX.

PR #502 (2018-12-15): Java11
Fixes #501 : Explains how to use Eclipse and OpoenJDK 11

Includes example changes (commented out) for .classpath on Windows.
closed 2018-12-15 by kasemir