ControlSystemStudio/phoebus #505: Splash screen site personalization

pavel-ch, 2018-12-17:
Is there any desire / intention to make the Splash Screen customizable like in Eclipse CSS?
I know it is possible to do some pre- or post-build solutions. The question is if it is worth of doing it more straightforward, for example by a new setting with png bitmap resource name. Resource will be supplied by the site product jar.
Thanks.

pavel-ch, 2018-12-17:
Now I found that creating the resource with the same name in my site product works for me. But I do not know how stable such a solution is. It works from command line (Maven build), but not from IDE. Maybe depends on the resource lookup order?

kasemir, 2018-12-17:
Right now, the way to change the splash in your site-specific product would be to replace the file phoebus/core/ui/src/main/resources/icons/splash.png in a local pre-build step.

As to why the replacement may work if you just put a resource with the same name into your product:
The Splash class uses Splash.getResourceAsStream("/icons/splash.png").
In principle, that would be the image inside the core-ui JAR where the Splash class resides.
But we currently create the product such that the class path is simply the list of all the jar files.
So it's one big name space, and you could end up with any "/icons/splash.png" in any jar file.

The same applies to our ImageCashe.  In principle it takes a class and a resource path relative to that class, but for now the class doesn't really matter because we have one big class loader.
Plus the ImageCache, well, caches. If we load image "x.png" from class A, and later try to load "x.png" from class B, the cache just checks the name and decices it already has a copy of "x.png", so you'd get the one from class A even if in the future class B would supply a different image.

==> The image names must be unique across all the phoebus source code.

Back to the splash,  the Launcher class currently has command line options `-splash` and `-nosplash`.
You could change `-splash` to take the path to some image, and then check that in the Splash class.

kasemir, 2018-12-17:
In fact, I'll add that, stay tuned..

kasemir, 2018-12-17:
Actually, you probably don't want a command line arg. After all, users who download your site-specific product should see your splash screen without need to call phoebus with "-splash /path/to/the/file.png".

Will look for a site_splash.png in the install location, meaning that file simply needs to be in the ZIP file at top-level.

pavel-ch, 2018-12-18:
Thank you, simple and effective.

closed 2018-12-18 by pavel-ch

kasemir, 2018-12-18:
See SNS example, https://github.com/kasemir/phoebus-sns/commit/0034d9cc48355ebf0735caa28f923ee9dfdca776,  for adding a site_splash.png into the zip file, so when unpacked, it's in the install location.

kasemir, 2018-12-18:
If you want to test this during development within Eclipse, I used Help -> About to find what it uses as the "Install Location". It was the git repo folder, so I copied a test site_splash.png there, but need to take care that you don't check that in.