ControlSystemStudio/phoebus #508: Multimonitor setups & full screen

pavel-ch, 2018-12-18:
The FullScreen action in Display Builder runtime works well on one monitor, however, I have identified some issues with that (all observations done on Windows):
1. In a multimonitor setup, it is not possible to fill automatically all monitors with Phoebus (an usual mode for our SCADA apps):
    a) Having detached windows / stages per monitor, only one of them can be made "full Screen" - JavaFx limitation, see 
        https://stackoverflow.com/questions/13030556/multiple-javafx-stages-in-fullscreen-mode

    b) Having one main window only and using splits on monitor borders - another possible way,  but setting this to "full Screen" limits the window back to one monitor. The same does maximizing the window.

Desired state: the Phoebus application covering the whole desktop using an "extended" full screen action, via method a) and/or b).

I can do investigation in this direction, however, this will need some additions to the core code.
So please let me know if such a functionality is wanted also from your side.
If not, that can be a show stopper for me , because this is something I can hardly implement inside the site product.

Minor remarks:
2. The Full Screen Action affects the whole Phoebus window (current stage) including all contained application tabs. As such, it does not logically belong to the Display Builder Runtime context menu,  but rather to the global menu/toolbar level.

3. The Full Screen action should be under authorization control (like locking a dock pane)

kasemir, 2018-12-18:
Can you start by moving the action from the display runtime to the window menu?
Then add the authorization control?

Those are both good points, I agree it would be better as you suggest.

As to how the full screen then is actually implemented, I'd have to see how that functions on Linux.
So maybe do that last.

pavel-ch, 2019-11-01:
FullScreen action - done:
- moving the action from the display runtime to the window menu
- authorization control
The FullScreen for multimonitor:
I do not see any suitable solution in Phoebus implementation.
So closing this issue.

closed 2019-11-01 by pavel-ch