ControlSystemStudio/phoebus #513: Alarm Config Logger : Push to remote git fails due to no upstream branch set

tanviash, 2018-12-20:
Hi Kunal,

I still see the problem for config-logger where if upstream is not set from local repo, push to remote doesn't happen.

shroffk, 2018-12-26:
Did you see anything in the logs?
In the above commit I added a bit more logging for when the remote is configured...can you check it out to see if there is some more info on why the push is failing.

tanviash, 2018-12-27:
Nothing in the logs related to this which means  git.remoteAdd().setName(REMOTE_NAME).setUri(uri).call() does not fail. When I try to push from local repo, this is what is seen.
deployer@phoebus-alarm-dev:/var/lib/AlarmConfig/Accelerator$ git push master
fatal: The current branch master has no upstream branch.
To push the current branch and set the remote as upstream, use

    git push --set-upstream master master

Take a look at this link which talks about similar issue with jgit library https://stackoverflow.com/questions/27823940/jgit-pushing-a-branch-and-add-upstream-u-option

tanviash, 2018-12-27:
This is .git/config looks like when in failed state.
deployer@phoebus-alarm-dev:/var/lib/AlarmConfig/Accelerator$ cat .git/config 
[core]
	repositoryformatversion = 0
	filemode = true
	logallrefupdates = true
[remote "remote"]
	url = https://git.nscl.msu.edu/alarm-config-logger/phoebus-alarm-config-test.git
	fetch = +refs/heads/*:refs/remotes/remote/*

shroffk, 2018-12-27:
after deleting the local repo, were you able to get it to work...can you run the service in debug mode. 
Put a break point at the pushCommand.call()

closed 2018-12-27 by tanviash