ControlSystemStudio/phoebus #514: Alarm Logging UI: Missing dependency and preference setting error

tanviash, 2018-12-20:
Alarm Logging-UI preference settings for es_host doesn't work. Also, logging-ui package as a dependency is missing from phoebus-product/pom.xml.

kasemir, 2019-01-04:
It's currently easy to remove functionality from a product (delete the jar file) but a little harder to add (need to build the jar file, add it, _and_ add that jar file to the classpath products manifest).
That's why I include the alarm UI in the SNS product, and then delete the jars in most installations.
I'm OK with similarly including the alarm UI in the default product.

tanviash, 2019-01-04:
@shroffk Let me know if we are adding the alarm-logging-ui dependency to default product as Kay mentioned above or going the individual site specific way.

shroffk, 2019-03-19:
This has been added.

closed 2019-03-19 by shroffk