ControlSystemStudio/phoebus #518: Alarm logger: Detect when alarm latches

kasemir, 2019-01-07:
The alarm logger captures the alarm `severity` as well as the `current_severity` of the PV.

This makes it easy to report how often a PV goes into alarm (`current_severity`), but it's hard to report how often a latched alarm triggered because that requires tracking the `severity` and detecting when it changes.

One attempt looks like this, using map/reduce scripts to track the severity changes:
```
GET /ihc_alarms_state_*/_search
{
  "query" :
  {
    "bool":
    {
      "filter":
      {
        "range" :
        {
            "message_time" :
            {  "gt" : "now-1d",      "lte" :  "now"    }
        }
      }
    }
  },
  "sort": [
    {
      "message_time": {
        "order": "asc"
      }
    }
  ],
  
  "aggs":
  {
    "latched_alarms":
    {
      "scripted_metric":
      {
        "init_script": """
          params._agg.last = new HashMap();
          params._agg.count = new HashMap();
        """,
        
        "map_script": """
          // Called for each document within a shard
          String pv = doc.pv.value;
          String sev = doc.severity.value;
          if (! params._agg.count.containsKey(pv))
          {
            // New alarm
            params._agg.count.put(pv, 1);
            params._agg.last.put(pv, sev);
          }
          else if (! sev.equals(params._agg.last.get(pv)))
          {
            // Latched severity changed.
            params._agg.last.put(pv, sev);
            // Count if it's an alarm
            if (!sev.equals("OK") && !sev.endsWith("ACK"))
              params._agg.count.put(pv, params._agg.count.get(pv) + 1);
          }
        """,
        
        "reduce_script": """
          // Called with array of shard results
          def last = new HashMap();
          def count = new HashMap();
          for (a in params._aggs)
            if (a != null)
              for (pv in a.count.keySet())
                if (! count.containsKey(pv))
                { // First shard that has counts for that PV
                  count.put(pv, a.count.get(pv));
                  last.put(pv, a.last.get(pv));
                }
                else
                { // Other shard had counts for that PV.
                  // Did that shard end with the same severity?
                  if (! last.get(pv).equals( a.last.get(pv) ))
                  { // Different last severity, include in count
                    count.put(pv, a.count.get(pv));
                    last.put(pv, a.last.get(pv));
                  }
                }
          return count;
        """
      }
    }
  }
}
```

A much easier approach has the alarm-logger populate a new boolean "latch" field which indicates when an alarm latches, so Kibana can then simply look for messages where latch=true:

[image]

kasemir, 2019-01-09:
The 'latch' is now detected in the alarm logger.
Have alarm server publish 'latch' in kafka message so that alarm logger doesn't need to figure this out. Might also be useful for other clients that only care about the time when an alarm first latches.

Deferring this change to ~early Feb.

shroffk, 2019-03-14:
can we close this?

closed 2019-03-14 by kasemir