ControlSystemStudio/phoebus #564: logConfiguration application does not appear in menu

pavel-ch, 2019-02-26:
Wrong spi service name:
/app-logConfiguration/src/main/resources/META-INF/services/org.phoebus.framework.spi.MenuEntry
instead of:
/app-logConfiguration/src/main/resources/META-INF/services/org.phoebus.ui.spi.MenuEntry

Should I send a PR or maybe faster you fix it by yourself?

br Pavel

kasemir, 2019-02-26:
Sure.
Related: Does it actually work for you?
For example, at a FINE logging level for org.phoebus.pv, you should see how each channel connects, receives updates etc. But I haven't figured out how to configure the log configuration UI to accomplish that. Similarly, the phoebus Launcher has a command line option `-logging` to read the log manager configuration from a file. But that doesn't have any effect for me.
The only changes that really work for me are the log level settings in phoebus/phoebus-product/src/main/resources/logging.properties.

shroffk, 2019-02-26:
I thought we had tested this once Kay, where we modified the logging levels for the requests library and saw the appropriate changes.
I can test that again next week.

kasemir, 2019-02-27:
You're right, I remember we tried this during a hangout, but when I last used the GUI or the `-logging ..` cmd line option, that had no effect. The might be related to changing the log level for a sub-logger to FINE while the top-level logger or handler is still at WARNING and thus blocks the more detailed messages.

pavel-ch, 2019-02-27:
For me, it was useful for eclipse paho logging (I use MQTT and the paho lib, by default, logs at FINE level to console, effectively making it impossible to see any other relevant output from my other debugs). Using this app, it was possible to switch paho logs off. I did not tested cmdline option, but I do because that can help, too.

pavel-ch, 2019-02-27:
For me, it appers to be working also from the command line.

kasemir, 2019-03-04:
Not sure what happened when I last tried it, but right now with your commit merged, I can 
 * Open Applications, Utility, Logging Config
 * In there, locate the org.phoebus.pv entry
 * Set its Logging Level to "All"
.. and now I get detail about each received value on the console.
So works fine, thanks!

closed 2019-03-04 by kasemir

pavel-ch, 2019-04-09:
The original error has been introduced again by #590, sending PR again