ControlSystemStudio/phoebus #575: Cannot update to JavaFX 11.0.2

kasemir, 2019-03-15:
JavaFX for 11 is now at 11.0.2, https://gluonhq.com/products/javafx/#ea.

But when I update the org.openjfx javafx-* versions in dependencies/phoebus-target/pom.xml (to fetch the target platform) and phoebus/core/ui/pom.xml (for the maven build),
that pulls in not only javafx-base-11.0.2.jar, javafx-controls-11.0.2.jar etc. as expected but also javafx.base-11.0.0-SNAPSHOT.jar

I can fix that in the ant build by simply deleting the older jars, but no idea how to fix the maven build to only use version 11.0.2.

shroffk, 2019-03-20:
I just update to the latest eclipse and am switching to openjdk11.0.2

Currently, I can build phoebus via maven...however in the IDE the javafx dependencies are not being resolved. I will keep you updated

kasemir, 2019-03-20:
When I tried it, maven built, but the product/lib contained javafx.base-11.0.0-SNAPSHOT.jar.
I assume that was dragged in by some other dependency.
Were you able to update everything so it's only JFX*11.0.2* jars in the product?

shroffk, 2019-03-20:
It is being pulled in by some other module...

I removed all the javafx modules...and built the target and it still had the snapshots.
```
    <dependency>
      <groupId>org.openjfx</groupId>
      <artifactId>javafx-controls</artifactId>
      <version>11.0.2</version>
    </dependency>
    <dependency>
      <groupId>org.openjfx</groupId>
      <artifactId>javafx-web</artifactId>
      <version>11.0.2</version>
    </dependency>
    <dependency>
      <groupId>org.openjfx</groupId>
      <artifactId>javafx-swing</artifactId>
      <version>11.0.2</version>
    </dependency>
    <dependency>
      <groupId>org.openjfx</groupId>
      <artifactId>javafx-fxml</artifactId>
      <version>11.0.2</version>
    </dependency>
```

shroffk, 2019-03-20:
Ok,
I found it...It is the springboot packages.
I will work more on this tomorrow.

shroffk, 2019-03-21:
The fix required the springboot version to be updated too

shroffk, 2019-03-21:
@kasemir is there a reason why the openjfx dependencies are of type pom?

kasemir, 2019-03-21:
Why does springboot require any UI dependency?

As for 'type pom', you should not ask me any maven questions, I have no clue.
I'm only typing 'maven clean install' to keep you happy.

shroffk, 2019-03-21:
> Why does springboot require any UI dependency?

I will add this to the questions for FRIB. I am hoping to hear from jeremy about the use of springboot in production deployment of services.

closed 2019-03-21 by shroffk