ControlSystemStudio/phoebus #581: Question about automated actions of an alarm

tanviash, 2019-03-19:
@kasemir Hello Kay, is there a way to use automated actions from 'Configure item' menu option of a PV in alarm tree to send emails to someone when PV goes in alarm state?

kasemir, 2019-03-19:
See online help file, https://github.com/shroffk/phoebus/blob/master/app/alarm/ui/doc/index.rst

Automated Actions

mailto:user@site.org,another@else.com: Sends email with alarm detail to list of recipients.

The email server is configured in the alarm preferences.

cmd:some_command arg1 arg2:
    Invokes command with list of space-separated arguments. The special argument "*" will be replaced with a list of alarm PVs and their alarm severity. The command is executed in the command_directory provided in the alarm preferences.

tanviash, 2019-03-20:
@kasemir Thanks Kay. I noticed a bug with automated actions. If I add an automated action to a PV, and then disable the alarm and re-enable it, the automated action goes away whereas it should have persisted on disabling the alarm.

kasemir, 2019-03-20:
Good that you found that.
The configuration for a PV or any node in the alarm config should always be a complete line in the Kafka config topic. I think the error is in  org.phoebus.applications.alarm.ui.tree.EnableComponentAction:

```
for (AlarmClientLeaf pv : pvs)
{
   final AlarmClientLeaf copy = pv.createDetachedCopy();
   if (copy.setEnabled(doEnable()))
              model.sendItemConfigurationUpdate(pv.getPathName(), copy);
}
```

Check createDetachedCopy(). It should create a complete copy of the PV, but I guess it's omitting the automated actions and maybe more, so those are removed when enabling/disabling a PV.

Can you fix that?

tanviash, 2019-03-20:
@kasemir I have a new pull request with the fix. Thanks.

closed 2019-03-20 by tanviash