ControlSystemStudio/phoebus #597: PV losing its configuration on re-starting phoebus

tanviash, 2019-03-22:
@kasemir Hi kay, With the latest code, when I re-start the phoebus application, any configuration that pv in the alarm tree has goes away. For instance, if PV was disabled, then after phoebus is restarted, PV comes up as enabled.

kasemir, 2019-03-25:
I cannot reproduce that.

 * Create new config
 * Add component 'Main'
 * Add PVs sim://sine and sim://ramp under Main
 * Verify that I get alarms, can acknowledge, alarm clears, alarm comes back
 * Disable sim://sine

Exit Phoebus, start phoebus back up: sim://sine is still disabled.
As expected, the Kafka config also still lists the PV as disabled:
```
sh monitor_topic.sh Accelerator
...
CreateTime:1553534292569: /Accelerator/Main/sim:\/\/sine: {"user":"ky9","host":"diane.ornl.gov","description":"sim://sine","enabled":false}
```
.. and the alarm server also lists it as such:
```
/Accelerator > ls -disabled  
  1 : /Accelerator/Main/sim:\/\/sine [sim://sine] - disabled - annunciating - latching - Current: OK/None (-2.9389262614623695), 2019-03-25 13:18:12.536, Alarm: MAJOR/HIHI (4.755282581475768), 2019-03-25 13:17:45.536
```

Restart alarm server, and new alarm server also has sim://sine as disabled.

tanviash, 2019-03-26:
@kasemir Thanks Kay. Yes it works. We were having some issues with the kafka setup that caused the symptoms I was seeing. I'll close this issue. Thanks again.

closed 2019-03-26 by tanviash

kasemir, 2019-03-26:
Well, that's good and bad to hear.
Happy that there's no new problem in the UI/client/alarm server.

With Kafka, the one issue that I cannot figure out is the log cleaner.
We create the config and state topics with `cleanup.policy=compact, ...` so that it only keeps the most recent message for a topic.
The `kafka/logs/log-cleaner.log` shows periodic compaction:
```
Log cleaner thread 0 cleaned log AcceleratorState-0 (dirty section = [0, 0])
...
Start size: 0.1 MB (414 messages)
End size: 0.1 MB (380 messages)
```

But that never happens for me with a fresh startup. When I start zookeeper & kafka with an empty setup, create the topics, start the alarm server & UI, configure it, ... the Kafka log cleaner never runs.
Once I stop & restart everything, the log cleaner runs OK.

shroffk, 2019-03-26:
Considering this issue is closed I might be too late.
But I failed to recreate this error on my side too. Restarting of phoebus did not result in any changes (enable/disable) or the alarm config.
The one thing I do see is that deleted/recent modifications appear and disappear upon quick restart...but this has been explained by Kay and not a major issue.

 > Once I stop & restart everything, the log cleaner runs OK.

I have invariably had to stop restart kafka and thus not personally observed or worried about this.