ControlSystemStudio/phoebus #601: Display Builder editor often shows "File has changed" dialog on Windows / NTFS

pavel-ch, 2019-03-27:
The dialog in `DisplayEditoInstance.doSave` method:
```
prompt.setTitle("File has changed");
prompt.setResizable(true);
prompt.setHeaderText(
  "The file\n   " + file.toString() + "\n" +
  "has been changed while you were editing it.\n\n" +
  "'OK' to save and thus overwrite what somebody else has written,\n" +
  "or\n" +
  "'Cancel' and then re-load the file or save it under a different name.");
```
is shown quite often without a real reason when saving a .bob file.
Seems to be a file write timing issue on the particular OS/Filesystem.
The explicit flush() on output streams seems to improve the situation for me, although it may not be a 100% solution. It should not harm anyway.
PR follows.

kasemir, 2019-03-27:
Well, calling `flush` twice, once inside `ModelWriter.writeModel()` and then in `EdiutorGUI.saveModelAs()`, is a bit like swinging a dead cat in the hope of better times to come.
Just writing and then closing the `Writer` should already do it.
We then fetch the `Files.getLastModifiedTime()` afterwards in the same thread, so that should be the actual file time.

Unless, as you already suggest, there's some buffering in the OS/Filesystem, but what can we do about this? Fetch the modification time after some delay? But then you wonder what delay would be appropriate...

Can you please try these:

a) Is `Files.getLastModifiedTime()` actually returning something useful, or do you get (Java Doc) "an implementation specific default when a time stamp to indicate the time of last modification is not supported by the file system"?

b) Maybe the issue is that we write with the `FileOutputStream` API, then check the modification date via the nio `Files` API. Can you replace the `modification_marker` with a long and use `File.lastModified()`? Maybe that reports a more consistent modification marker on Windows.

pavel-ch, 2019-03-27:
Fine.
ad calling flush() twice: it may be a dead cat but actually it helps to get rid of the false dialog occurrences.
ad a) the `Files.getLastModifiedTime()` is actually returning something useful, because it works on my system at around 90% retries. Also checked by debugs. But in some 10% it returns a time (after write) which is a bit earlier than on the later call before a new Save.
ad b) I give it a try

pavel-ch, 2019-03-28:
After some experiments:
The `File.lastModified()` API alone gives the same results like `Files.getLastModifiedTime()`. What makes the difference is the `flush()` call.
The `File.lastModified()` is maybe a cleaner solution, using the same API in one class. Should I PR?

pavel-ch, 2019-04-01:
Thank you for merging

closed 2019-04-01 by pavel-ch