ControlSystemStudio/phoebus #606: Ctrl+Drag and Drop does not copy widget in Display Editor

pavel-ch, 2019-03-29:
There is already a DragDrop support in editor. What should from my perspctive work, too, is a Ctrl-Drag and Drop gesture for the widget copy operation in editor.
At the moment, the drag from a widget returns always the same string:
```
<?xml version="1.0" encoding="UTF-8"?>
<display version="2.0.0">
</display>
```
And thus the copied widget cannot be instantiated based on this string.
I can try to find the issue but tracking all the DnD support takes too much time to me. Can I get an advice where to start?

kasemir, 2019-04-01:
When you select one or more widgets, the `Tracker` is what's showing the blue selection rectangle with little handles to move and resize.

Check `org.csstudio.display.builder.editor.tracker.SelectedWidgetUITracker`, line 124, `setOnDragDetected`, which checks for the start of a drag operation.

Normally, when you just press-left-button-and-drag, we intend to move the selected widgets, which is why `setOnDragDetected` bails out right away unless `event.isShortcutDown()`, meaning you must be holding Ctrl (or Command on the Mac) to start a drag-and-drop instead of just moving the tracked widgets around.

If you raise the log level or change this source line to see what widgets are selected, you'll find that none are when the ctrl-drag is started: 
```
         // Allow 'dragging' selected widgets
        setOnDragDetected(event ->
        {
            if (! event.isShortcutDown())
                return;

            logger.log(Level.FINE, "Starting to drag {0}", widgets);
```

I think that's because Ctrl-click is also used to toggle selection of individual widgets.
Say you select a bunch of widgets, you can then Ctrl-click to add/remove widgets to/from the selection.

So select a widget, Ctrl-hold-down to start dragging
==> The Ctrl-click de-selects the widget, and you drag nothing.

kasemir, 2019-04-01:
The handling of Ctrl-Click to toggle the selection of widgets starts in `org.csstudio.display.builder.editor.tracker.SelectedWidgetUITracker` line 117:
```
        // Pass control-click down to underlying widgets
        addEventFilter(MouseEvent.MOUSE_PRESSED, event ->
        {
            if (event.isShortcutDown())
                passClickToWidgets(event);
        });
```

and a stack trace from there on looks like this:
```
at org.csstudio.display.builder.editor.WidgetSelectionHandler.toggleSelection(WidgetSelectionHandler.java:76)
at org.csstudio.display.builder.editor.DisplayEditor$1.handleClick(DisplayEditor.java:367)
at org.csstudio.display.builder.representation.ToolkitRepresentation.fireClick(ToolkitRepresentation.java:580)
at org.csstudio.display.builder.editor.tracker.SelectedWidgetUITracker.passClickToWidgets(SelectedWidgetUITracker.java:355)
at org.csstudio.display.builder.editor.tracker.SelectedWidgetUITracker.lambda$3(SelectedWidgetUITracker.java:120)
```

So if you select more than one widget, and then use Ctrl-down to drag them, you are de-selecting the widget under the mouse and then you drag the rest.

The thing to fix here would be to come up with some way to allow all of the following without overlap in the used mouse buttons and modifier keys:

 * add/remove widgets from the selection
 * move the selected widgets
 * drag/drop the selected widgets

pavel-ch, 2019-04-01:
Okay understood.
Would it be acceptable the following scenario:
- if just one widget is selected (`SelectedWidgetUITracker.widgets.size()==1`), do not propagate Ctrl-click to the widget itself
- the solution works for me, the only side effect is that you cannot deselect a single selected widget with Ctrl-click.. From my understanding, this gesture has no practical usage.
Resulting code:
```
        // Pass control-click down to underlying widgets
        addEventFilter(MouseEvent.MOUSE_PRESSED, event ->
        {
            if (event.isShortcutDown() && widgets.size()!=1)
                passClickToWidgets(event);
        });
```

claudio-rosati, 2019-04-02:
> the only side effect is that you cannot deselect a single selected widget with Ctrl-click.. From my understanding, this gesture has no practical usage.

I'm sorry but I disagree. This makes the system asymmetrical. Moreover, on macOS copy is done pressing ALT+drag, and this is what a mac user is expecting.

What I would do is to change the event filter to listen for `MouseEvent.MOUSE_CLICKED` instead, and in the following `setOnDragDetected` statement look at the `event.getTransferMode()` and check whether it contains `TransferMode.COPY`.

pavel-ch, 2019-04-02:
Yes I have considered this solution in the meantime, too.
Seems to work well except there is no `event.getTransferMode()` method, so I do no know what you exactly mean with 
> look at the event.getTransferMode() and check whether it contains TransferMode.COPY

The basic thing changed for me was replacement of `MouseEvent.MOUSE_PRESSED` with `MouseEvent.MOUSE_CLICKED` in `SelectedWidgetUITracker.addEventFilter`.
Then, in Tracker.java:135, it was necessary to block Tracker movement when Control is pressed:
```
        tracker.setOnMouseDragged((MouseEvent event) ->
        {
            if (start_x < 0 || event.isControlDown())
                return;
            double dx = event.getX() - start_x,  dy = event.getY() - start_y;
......
```
As I have not not a Mac available, I cannot check, but I think your suggestion does not follow your statement 
> on macOS copy is done pressing ALT+drag, and this is what a mac user is expecting.

For me on Windows, however, is this solution quite fine.

kasemir, 2019-04-02:
>  on macOS copy is done pressing ALT+drag, and this is what a mac user is expecting.

Isn't that inconsistent? Generally, Mac uses Command instead of Control. Command-C to copy instead of Control-C etc.
The Event API has `isShortcutDown()` to handle this: On mac it checks for Command, on other platforms for Control.
Why does the drag-copy suddenly use Alt?

claudio-rosati, 2019-04-02:
That is how D&D work on macOS:

- Alt (Option) -> COPY
- Alt (Option) + Meta (Command)  -> LINK

kasemir, 2019-04-02:
OK, #611 now uses Alt-drag on Mac. Ctrl-drag on other platforms.

Toggle selection still via Ctrl-click on most platforms, Command-click on Mac.

claudio-rosati, 2019-04-02:
Yes. Perfect. Thank you.

closed 2019-04-02 by kasemir