ControlSystemStudio/phoebus #612: Question about running alarm-server as service

tanviash, 2019-04-02:
@kasemir Hi Kay, you have an example provided in app/alarm/examples for how to run alarm server as systemd service using procserv where you provide --logfile option to procserv. Is it possible for procserv to write the alarm-server logs to system logs (/var/log/syslog) instead of to a file provided by --logfile?

ExecStart=/usr/bin/procServ --foreground --noautorestart --name alarm-server --chdir ${SERVER} --logfile ${SERVER}/console.log ${PORT} ./alarm-server.sh -settings /PATH/TO/phoebus.ini -config ${CONFIG}

kasemir, 2019-04-03:
It's a procserv issue, moved there.

closed 2019-04-03 by kasemir

kasemir, 2019-05-15:
https://github.com/ralphlange/procServ/issues/25 suggests a way of running procServ such that it will log to syslog