ControlSystemStudio/phoebus #613: Diplay Editor - Copy/Paste Property values

pavel-ch, 2019-04-03:
Is there any intention to implement the Copy/Paste widget property values function for Display Editor?
The function is implemented in BOY and is handy, especially for rules/actions etc.

claudio-rosati, 2019-04-03:
It is on my TODO list (CSSTUDIO-743) but I can't tell you when I'll do it.
Currently what you can do is copying your "source" widget and then morphing it into the one you wish (context menu **Replace with**).

pavel-ch, 2019-04-03:
Fine thanks, I will wait a bit. If you feel you must postpone it too much, I can try to help.

kasemir, 2019-04-05:
You already have some of that.
When you select multiple widgets, the first widget that you select is special.

For example, select a widget that has a rule.
Then add several other widgets to the selection.
The property view now shows the common properties of all selected widgets, and under "Rules" you see that there is one, the one from the widget selected first.
"Edit" the rules, but don't do anything other than pressing "OK" to keep the rule unchanged.

--> Now all selected widgets have that rule.

Same with for example actions:
Select widget that has the action, then select others, "edit" actions by simply opening action dialog and pressing "Enter" 
--> Now all selected widgets have that action.

claudio-rosati, 2019-04-05:
I was unaware of that!
Anyway I think a bit difficult to remember it, and moreover it require a proper selection workflow.
Maybe having a couple  specific commands (copy & paste) will ease the transition from BOY, and they are visible (and not necessarily rememberable) on the context menu.

kasemir, 2019-04-05:
I agree that it's no replacement for copying/pasting selected properties, since that would support copying multiple properties at once.

Otherwise, I think it's pretty straight forward.
When you select multiple widgets, you see their common properties.
Like if they all support rules, actions, background color, font, ... you see those common properties.
When you set a color, font, rule, ... then naturally all selected widgets get assigned that color, font, rule.

The only debatable question:
If I select N widgets and their background colors are red, green, blue, yellow, ..., which background color should show up in the common property view?
a) The color of the widget selected first
b) The color of the widget selected last
c) The color created by combining the colors of all selected widgets (white? black?)

I can usually not remember if it's a) or b), but just checking the property panel while selecting the first, next, .. widget clarifies it.