ControlSystemStudio/phoebus #618: Prepare for epics 7.0.3

shroffk, 2019-04-03:
The latest snapshot releases of epics should be accessible in phoebus in order to test the new versions of epics core modules before they are released

shroffk, 2019-04-04:
@kasemir the branch Issue_618 has been setup to use the latest snapshot releases for epics 7.0.3-SNAPSHOT and vtype 1.0.1-SNAPSHOT.

In the pom files I have created properties for epics.version and vtype.version so that in the future we can easily manage versions in the future. However, since we manage the classpath files manually, the version numbers there will need to be manually managed.

I am not sure if this should be merged.

kasemir, 2019-04-04:
Thanks, I'll use that branch to see if the VTable hacks can be replaced, if the added Array classes can be removed etc. and then create a pull request.

kasemir, 2019-04-04:
Removing the VStats and the VArray classes from core-vtype mostly needed tweaks to statistics and enum code.

New VStats needs Display, which isn't always defined for the averaged data.

New VEnumArray wants ListInteger (ArrayInteger, ..). The Channel Access enum provides `short[]`. Using a ListNumber would allow ArrayShort as well as ArrayInteger, saving one array copy.

shroffk, 2019-04-04:
> New VEnumArray wants ListInteger (ArrayInteger, ..). The Channel Access enum provides short[]. Using a ListNumber would allow ArrayShort as well as ArrayInteger, saving one array copy.

+1 will add this

kasemir, 2019-04-04:
Alright. Otherwise looks good. I'm removing the core-vtype module which is now basically empty since TableHacks, VStats, V*Array classes are all in epics-vtype.
Should have a PR in an hour.

shroffk, 2019-04-05:
What should happen is someone provides a float[] or double[] to VEnumArray.

kasemir, 2019-04-05:
I suggest accepting any ListNumber, reading it via getShort() or getInt().
So floating point data will be truncated.

shroffk, 2019-04-05:
I think I have a workaround
https://github.com/epics-base/epicsCoreJava/commit/8d949ca07eed7bb4c4a831ba6448d4a67c35a4d5

The VEnumArray uses ListNumber but the factory method only allows ListInteger and ListShort. This way we can support both short[] and int[] awhile discouraging the use of other.

Additionally, 
Just like with VEnum, I had removed the exception for index out of bound...instead when you have a index out of the enum range the label reads "invalid index <value> ...."

kasemir, 2019-04-05:
Yes, good! Will that soon appear in vtype-1.0.1-SNAPSHOT, or by some other way into phoebus?

shroffk, 2019-04-05:
I have deployed the latest changes to the snapshot repo already.

A new build of the dependencies should get the latest version in phoebus.

kasemir, 2019-04-08:
Alright, I see it. Thanks!

closed 2020-06-04 by shroffk