ControlSystemStudio/phoebus #620: Scaled Slider increment ignored?

kingspride, 2019-04-04:
Hi,

is it possible to change the scaled slider in a way that it doesn't jump directly to the clicked location, but instead obeys the increment setting and only changes by increment size in the clicked direction?

I'm asking because my facility is currently converting old opi files and CSS into bob files and phoebus. from the old environment, they are used to clicking somewhere left or right from the sliders knob to change the value in (small) steps.

what I want:
[image]

what phoebus currently does:
[image]

shroffk, 2019-04-04:
Could this be a widget property so that the behaviour can be configured by the OPI designer.

kasemir, 2019-04-04:
This behavior is hardcoded inside the JavaFX Slider, i.e. in the graphics library that we use with the Display Builder, no matter if Eclipse/RCP or Phoebus.
It's this code inside `com.sun.javafx.scene.control.behavior.SliderBehavior`:
```
    /**
     * @param position The mouse position on track with 0.0 being beginning of
     *        track and 1.0 being the end
     */
    public void trackPress(MouseEvent e, double position) {
        // determine the percentage of the way between min and max
        // represented by this mouse event
        final Slider slider = getNode();
        // If not already focused, request focus
        if (!slider.isFocused()) slider.requestFocus();
        if (slider.getOrientation().equals(Orientation.HORIZONTAL)) {
            slider.adjustValue(position * (slider.getMax() - slider.getMin()) + slider.getMin());
        } else {
            slider.adjustValue((1-position) * (slider.getMax() - slider.getMin()) + slider.getMin());
        }
    }
```

I can even argue that's a good idea, because that way a click has a predictable response: The slider moves to the clicked position.
If a click-and-hold results in a sequence of updates, how fast the slider moves really depends on how busy the computer is at the time, plus you create one 'write' to the PV per update, and when each written value reaches the IOC and takes effect is yet another question.

I do understand the desire to adjust some value in "steps" while interactively monitoring the response.
With the slider, that's possible by pressing the left/right respectively up/down cursor keys. Each key press adjusts the slider by the 'increment' value which can be configured on the widget in edit mode or at runtime via the context menu.

kingspride, 2019-04-04:
well using the keyboard doesnt really work out since you have to click on the slider to get the focus. which already can inadvertently move it by a small amout.
(I think we all know missclicking on small UI elements)

From your answer, I can't really tell if you're willing to change something to get my wish done not - if not, no problem. I think there are enough other controls to replace that slider adequately. Just clarify , please :)

kasemir, 2019-04-04:
>  I can't really tell if you're willing to change something to get my wish done not 

What I tried to explain is that the Slider widget that we use simply works this way. The underlying widget has no option to step-step-step while the mouse button is held down.
To get it to do what you want, we'd have to replace the normal JavaFX Slider widget with a different one.
Or you can start an issue ticket with open jfx to change the behavior of the JavaFX Slider, https://github.com/javafxports/openjdk-jfx

kasemir, 2019-11-06:
A few version updates of JavaFX later, the Slider still doesn't work that way.
Ideally, com.sun.javafx.scene.control.behavior.SliderBehavior` would be public API, since in there it's easy to fix. But found a way to capture mouse clicks and get the desired behavior with basically one page of code, so not too horrible.

georgweiss, 2019-11-07:
Agree with @shroffk , if implemented, let's make the wanted behavior optional through a property.

kasemir, 2019-11-07:
It's now a preference, where the default behavior is the same as EDM and BOY, as original ticket requested.
I think it would be misleading to have a widget property for this behavior, because then users end up with screens where identically looking sliders can behave in different ways.
With a preference setting, each site can decide which behavior they want for their sliders on all their displays.

kasemir, 2019-11-14:
Closing, requested inc/dec behavior has been implemented, with preference to select jump-on-click.

closed 2019-11-14 by kasemir