ControlSystemStudio/phoebus #623: Assigning different PVs to different parts of the display causes some race condition

SeatherMarqx0, 2019-04-04:
Getting more than one PV from the same server and assigning them to different visual elements results in one or both of them disconnecting.

Phoebus then cannot probe either of the values desired, but `pvget` works.
Refreshing the page sometimes swaps which visual element gets updated/connected.
Closing the page allows probing of individual PV elements again.

I have two records that look like the following:
```
test:Monit:localhost:Service:sshd structure 
    time_t timeStamp 2019-04-04 13:39:23.059  
        long secondsPastEpoch 1554410363
        int nanoseconds 59000000
        int userTag 0
    epics:nt/NTScalar:1.0 name
        string value sshd
    epics:nt/NTScalar:1.0 enabled
        int value 1
    epics:nt/NTScalar:1.0 state
        int value 0
    epics:nt/NTScalar:1.0 running
        int value 1
        alarm_t alarm 
            int severity 0
            int status 0
            string message 
    epics:nt/NTScalar:1.0 type
        string value PROCESS
```
```
test:Monit:localhost structure 
    time_t timeStamp 2019-04-04 13:41:23.306  
        long secondsPastEpoch 1554410483
        int nanoseconds 306000000
        int userTag 0
    alarm_t alarm 
        int severity 0
        int status 0
        string message 
    epics:nt/NTScalar:1.0 host
        string value localhost
    epics:nt/NTScalar:1.0 pollRate
        int value 120
        display_t display
            double limitLow 0
            double limitHigh 0
            string description 
            string format 
            string units S
```

Error occurs when more than one element is looking at different PVs.
Example:
2 TextUpdate elements.
One displaying <pva://test:Monit:localhost:Service:sshd/name>
The other <pva://test:Monit:localhost/host>

System Info:
Phoebus commit [59e1073b7b56f93546538c715a4d23f5109544b7](https://github.com/shroffk/phoebus/commit/59e1073b7b56f93546538c715a4d23f5109544b7)
Phoebus Build with Oracle JDK 11
Server written in Java (built with Oracle JDK 11).

Server EPICS dependencies (from Maven repo):
```Groovy
compile group: 'org.epics', name: 'epics-core', version: '7.0.2'
compile group: 'org.epics', name: 'epics-pvdatabase', version: '4.3.1'
compile group: 'org.epics', name: 'epics-pvdata', version: '6.1.2'
```

kasemir, 2019-04-05:
Yes, this is a known bug in the pva:// support. Thank you for creating a ticket.
Unclear if the problem is in the way we use the pva library, or in the Java pva library itself, since I don't remember seeing this problem with the RCP-based Display Builder that used an earlier version of the Java pva library.

kasemir, 2019-04-09:
Simple demo runs this in ` softIocPVA`:
```
record(calc, "ramp")
{
  field(SCAN, "1 second")
  field(INPA, "ramp")
  field(CALC, "A+1")
}
record(calc, "rnd")
{
  field(SCAN, "1 second")
  field(CALC, "100*RNDM")
}
```

A display for both `pva://ramp` and `pva:rnd` might work when executed the first time, but on reload, or when accessing the PVs with probe, PV tree etc. results in not all PVs connecting.

This works fine with the RCP-based DisplayBuilder.

kasemir, 2019-04-09:
Distilled it down to example Java code that's totally unrelated to Phoebus, Display Builder, ..

Runs OK with older PVA library, but has connection errors with the latest PVA library.
Reported there: https://github.com/epics-base/epicsCoreJava/issues/53

wcork, 2019-04-24:
Latest pva libs in phoebus build seems to resolve this issue.

For the record, this MR seemed to have fixed it.
#641

closed 2019-04-24 by wcork