ControlSystemStudio/phoebus #624: Alarm Annunciator showing strange behavior

tanviash, 2019-04-04:
When I open the alarm annunciator while the sim PVs are in alarm state, I do not see them getting annunciated on annunciator table. Only if I acknowledge the PV, it starts to show up in the annunciator table.

kasemir, 2019-04-05:
Is this what you see?

 1. sim:// goes in and out of alarm. One such alarm has latched, so PV is shown in the alarm table.
 2. Open annunciator. Annunciator only annunciates new alarms, not those that are already active.
 3. Acknowledge sim:// PV. Alarm is removed from the alarm table.
 4. sim:// PV goes back into alarm. Alarm latches, is shown in alarm table. Since this is a new alarm, it's now annunciated.

==> That would be normal behavior.

tanviash, 2019-04-05:
Thank you Kay.

closed 2019-04-05 by tanviash