ControlSystemStudio/phoebus #629: Alarm Notifier Behavior in Maintenance Mode

tanviash, 2019-04-08:
@kasemir Hi Kay, under maintenance mode when automated action mailto: is set for a PV, email notifications are received for minor & major alarms. Is this an expected behavior for Phoebus maintenance mode? For cs-studio, automated actions under maintenance mode are suppressed for minor & major alarms. CC'ing @shroffk and @berryma4 .

kasemir, 2019-04-08:
I'd say yes.
Maintenance mode is only about Invalid/Undefined alarms, which are automatically acknowledged because in maintenance periods it's quite natural that IOCs are rebooted, and we don't want the Invalid alarms from that.
Other alarms should be handled as usual. If they weren't with the previous implementation, that's a bug in the previous implementation.

shroffk, 2019-04-08:
+1
I agree with Kay, even in maintenance mode...if your pv is available and reporting an alarm state then the alarm server should process it as normal.

tanviash, 2019-04-08:
@kay In this test document https://github.com/shroffk/cs-studio/blob/master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast.notifier/demo/doc/STP_CODAC_Alarm_Automated_Actions.pdf, in section 3.4.8 B1, it expects no major alarms under maintenance mode.

This feature is utilized by operators here to disable automated actions. Is there another way to disable automated actions in phoebus?

tanviash, 2019-04-08:
Thanks all. We can handle this a layer up by using mailing lists. Closing this issue.

closed 2019-04-08 by tanviash