ControlSystemStudio/phoebus #635: Display Editor - Support for Group locking

pavel-ch, 2019-04-12:
The locking of Group objects should be implemented in Display Editor, just like in Boy, to prevent unwanted contents changes in finished groups.
Any plans?

kasemir, 2019-04-12:
I've personally never used that feature. On the contrary, I find it bothersome when the original author of a display had locked groups so I need to unlock to edit.
Consequently, implementing this for the display editor is way down on my list of priorities.

claudio-rosati, 2019-04-12:
I agree with Kay.
What I have in my plan is to add on the Overview tree a visible/invisible "eye" button (linke in Photoshop layers) to temporarily set hidden some widgets (groups too) in order to work easily and safely on other part of the OPI. This invisibility status is temporary and never saved in the `.bob` file.

pavel-ch, 2019-04-12:
We often use a pattern of composing complex objects into groups and reusing such groups. These groups are typically transparent without any decoration. Such groups tend to be accidentally changed in the editor, when somebody wants to for example move the group but selects a group "child" widget instead.
This behavior can be described as that a click selects any widget independently of its group membership.
Other graphic editors face this situation for example:
- Group cannot be edited at all, must be dissolved and grouped again (the worst solution, losing group metadata)
- Lock / Unlock (Boy)
- Group must be somehow set to an "edit" mode first, for example:
      - First click selects group and allows manipulation with whole group, next click to the selected group selects member widget (Visio)
      - Group must be double-clicked to enter edit mode, double click outside to exit the mode (Inkscape)
Maybe another models can be found.
You can see that because it is even somehow implemented in other editors, the function has some usefulness.
I can live with the current implementation, but I think there are patterns which can be followed.

pavel-ch, 2019-05-22:
Ok see no support / closing. Can easily live with the current implementation.

closed 2019-05-22 by pavel-ch