ControlSystemStudio/phoebus #638: Phoebus client not updating bootstrap server

tanviash, 2019-04-17:
@kasemir Hi Kay, I have recently moved the alarm server & other services to use a new kafka cluster. When I start the Phoebus client with settings.ini file updated to use the new kafka cluster, the phoebus client logs still show the bootstrap server set to the older kafka cluster and eventually phoebus client can't connect to alarm server. 

I am running phoebus client as java -jar phoebus-product/target/product-0.0.1-SNAPSHOT.jar -settings ~/settings.ini where settings.ini file contains org.phoebus.applications.alarm/server set to the new kafka server.

tanviash, 2019-04-17:
The memento files inside .phoebus still had the old value for bootstrap server. The problem goes away if I delete .phoebus directory from home.

kasemir, 2019-04-18:
Hmm, that’s still a problem. Reading the ‘-settings’ should replace the preferences saved in~/.phoebus.
In fact just running with ‘-settings’should replace them, so from then on you would not need the command line option - though won’t hurt, either.

Could it be that the .Phoebus files were read-only so ‘-settings’ couldn’t write the preferences?

tanviash, 2019-04-25:
Closed by mistake, issue still exists where .phoebus directory has to be cleaned at times for new settings to start working

shroffk, 2022-04-08:
@tanviash I was unable to reproduce this issue on my windows development machine.
Is this still an issue?

tanviash, 2022-04-08:
Not an issue anymore.

closed 2022-04-08 by tanviash