ControlSystemStudio/phoebus #640: Window Menu glitch under x2go

mdavidsaver, 2019-04-18:
Running phoebus remotely using x2go mostly works.  However, I see an oddity wrt. window menus only (not context menus).

When I click on eg. 'File', the menu appears, but below the row of toolbar buttons.  I am then unable to select items by mouse.  Keyboard up/down/enter still works, so it is usable for me.

[image]

Details:

client is Debian 9.  X2Go Client v. 4.0.5.2 (Qt - 4.8.7)
server is CentOS 7.  server 4.1.0.3, agent 3.5.99.17

mdavidsaver, 2019-04-18:
I initially noticed this error printed on start:

```
ES2 Prism: Error - reported GLX version = 1.2
    GLX version 1.3 or higher is required
```

Adding `-Dprism.order=sw` makes this go away, but the menu glitch remains.

shroffk, 2019-04-18:
This has been reported already here:

https://github.com/shroffk/phoebus/issues/436#issuecomment-444231965

Reponse:

> Request for context menu in display builder was an actionable item which I could handle.
> Menu behaviour with a specific Linux setup is very different, more like #256 or #353 or #367 .
> Still, any difference when using -Djdk.gtk.version=2 vs -Djdk.gtk.version=3?
> I'm using GTK2 because otherwise Drag & drop doesn't work, i.e. hard to arrange tabs or edit displays.
>

mdavidsaver, 2019-04-18:
Ok.  my quick search didn't find this.

mdavidsaver, 2019-04-18:
Ah.  The menus look ok with `-Djdk.gtk.version=2` (from #367)

Sigh... I was so hoping that I could stop caring about GTK2 vs. 3.

mdavidsaver, 2019-04-19:
Never mind.  It still happens sometimes with gtk2.

@shroffk Do you know if there is an issue on openjfx for this?  The closest I can find is https://bugs.openjdk.java.net/browse/JDK-8176844 which claims to be fixed.

mdavidsaver, 2019-04-23:
@hhslepicka provides some insight.

Some of the mouse issues go away if the phoebus window loses, then regains, focus.  The remainder seem to exist with both GTK2 and 3.

Hugo saw similar problems with Qt on Mac, which were apparently fixed with a Qt upgrade (no GTK involved).

mdavidsaver, 2026-11-02:
In recent years I have mostly switched to running phoebus locally, and using either a VPN or PVA gateway + SSH port forwarding.  I will reopen and update if/when I encounter this issue again.

closed 2026-11-02 by mdavidsaver