ControlSystemStudio/phoebus #642: Using Archiver Appliance in Data browser is not working

DanielALS, 2019-04-19:
I have a settings.ini file with this:

```
# Databrowser preferences
org.csstudio.trends.databrowser3/archives=Current|0|pbraw\://localhost:17668/retrieval
org.csstudio.trends.databrowser3/urls=pbraw\://localhost:17668/retrieval
org.csstudio.trnds.databrowser3/update_period=0.5
org.csstudio.trends.databrowser3/scroll_step=1
```

When I open a data browser window and ask for data in the past, nothing comes up. There's no obvious errors.

Screen shot:
[image]

kasemir, 2019-04-20:
No idea about the AA, and may be unrelated, but why is there a `\` in `pbraw\://localhost...`?

shroffk, 2019-04-22:
The backslash was added cause we were getting some URL parsing exceptions...I can verify this again.

@DanielALS was this working before? Are you able to log into the AA admin page and confirm if the archiver is properly archiving this pv

DanielALS, 2019-04-23:
Hello guys,
I have not gotten the AA support in the display to work at all. I'm
positive that the PV is being archived.
Thanks

forestmojo, 2019-04-30:
in my setting.ini:
#archappl
org.csstudio.trends.databrowser3/urls=pbraw://localhost:17668/retrieval
org.csstudio.trends.databrowser3/archives=pbraw://localhost:17668/retrieval
it works.

shroffk, 2019-05-09:
@DanielALS I am assuming this issue is resolved

closed 2019-05-09 by shroffk