ControlSystemStudio/phoebus #644: Phoebus client throws error with multiple bootstrap servers

tanviash, 2019-04-24:
I can successfully use comma-separated kafka servers in the config for alarm server, alarm logger & alarm config logger, but when I try to use the same comma-separated list of kafka servers for Phoebus client through settings.ini, client throws error.
org.phoebus.applications.alarm/server=kafka01-dev.nscl.msu.edu:9092,kafka02-dev.nscl.msu.edu:9092,kafka03-dev.nscl.msu.edu:9092

15:48:22.682 [JavaFX Application Thread] DEBUG org.apache.kafka.clients.consumer.KafkaConsumer - [Consumer clientId=consumer-1, groupId=Alarm-bf1ed2c7-3e23-4916-a201-805ed1e90bd2] Initializing the Kafka consumer
15:48:22.774 [JavaFX Application Thread] WARN org.apache.kafka.clients.ClientUtils - Removing server null:9092 from bootstrap.servers as DNS resolution failed for null
15:48:22.777 [JavaFX Application Thread] DEBUG org.apache.kafka.clients.consumer.KafkaConsumer - [Consumer clientId=consumer-1, groupId=Alarm-bf1ed2c7-3e23-4916-a201-805ed1e90bd2] Kafka consumer has been closed
2019-04-24 15:48:22 WARNING [org.phoebus.applications.alarm] Cannot create alarm table for alarm://kafka01-dev.nscl.msu.edu:9092,kafka02-dev.nscl.msu.edu:9092,kafka03-dev.nscl.msu.edu:9092/alarm_test
org.apache.kafka.common.KafkaException: Failed to construct kafka consumer
	at org.apache.kafka.clients.consumer.KafkaConsumer.<init>(KafkaConsumer.java:799)
	at org.apache.kafka.clients.consumer.KafkaConsumer.<init>(KafkaConsumer.java:650)
	at org.phoebus.applications.alarm.client.KafkaHelper.connectConsumer(KafkaHelper.java:71)
	at org.phoebus.applications.alarm.client.AlarmClient.<init>(AlarmClient.java:74)
	at org.phoebus.applications.alarm.ui.table.AlarmTableInstance.create(AlarmTableInstance.java:76)
	at org.phoebus.applications.alarm.ui.table.AlarmTableInstance.<init>(AlarmTableInstance.java:46)
	at org.phoebus.applications.alarm.ui.table.AlarmTableApplication.create(AlarmTableApplication.java:59)
	at org.phoebus.applications.alarm.ui.table.AlarmTableApplication.create(AlarmTableApplication.java:51)
	at org.phoebus.applications.alarm.ui.table.AlarmTableApplication.create(AlarmTableApplication.java:24)
	at org.phoebus.framework.workbench.ApplicationService.createInstance(ApplicationService.java:142)
	at org.phoebus.applications.alarm.ui.table.AlarmTableMenuEntry.call(AlarmTableMenuEntry.java:45)
	at org.phoebus.applications.alarm.ui.table.AlarmTableMenuEntry.call(AlarmTableMenuEntry.java:21)
	at org.phoebus.ui.application.PhoebusApplication.lambda$createMenuItem$23(PhoebusApplication.java:527)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:49)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.control.MenuItem.fire(MenuItem.java:465)
	at com.sun.javafx.scene.control.ContextMenuContent$MenuItemContainer.doSelect(ContextMenuContent.java:1380)
	at com.sun.javafx.scene.control.ContextMenuContent$MenuItemContainer.lambda$createChildren$12(ContextMenuContent.java:1333)
	at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:218)
	at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
	at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$MouseHandler.process(Scene.java:3851)
	at javafx.scene.Scene$MouseHandler.access$1200(Scene.java:3579)
	at javafx.scene.Scene.processMouseEvent(Scene.java:1849)
	at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2588)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:397)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:295)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$2(GlassViewEventHandler.java:434)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:390)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:433)
	at com.sun.glass.ui.View.handleMouseEvent(View.java:556)
	at com.sun.glass.ui.View.notifyMouse(View.java:942)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$11(GtkApplication.java:277)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: org.apache.kafka.common.config.ConfigException: No resolvable bootstrap urls given in bootstrap.servers
	at org.apache.kafka.clients.ClientUtils.parseAndValidateAddresses(ClientUtils.java:66)
	at org.apache.kafka.clients.consumer.KafkaConsumer.<init>(KafkaConsumer.java:709)
	... 54 more

kasemir, 2019-04-24:
` Cannot create alarm table for alarm://kafka01-dev.nscl.msu.edu:9092,kafka02-dev.nscl.msu.edu:9092,kafka03-dev.nscl.msu.edu:9092/alarm_test`

How does the "alarm://" and "/alarm_test" get added to the list of `server:port`?

tanviash, 2019-04-24:
I'm not sure about alarm:// but alarm_test is the kafka topic name.

anantianshi, 2019-08-30:
Hello，tanviash！When I config the Alarm system，I have the same problem. The problem is refer to the following pipcture 
[image],Once the problem occurs,It will exist all the time. And also I can connect the kafka server just in localhost. The following picture is my setting.ini of phoebus.
[image]
.I do not know how to solve those problems.

kasemir, 2019-08-30:
> can connect just in localhost

I'm using this to connect to a non-localhost Kafka instance, and it works fine in production for months, so I don't see a fundamental reason that would limit it to localhost:
```
org.phoebus.applications.alarm/server=name_of_server:9092
```

If you have connection problems, I'd expect to see more detailed error messages on the console from where you start phoebus.

kasemir, 2019-08-30:
Have you tried without "PLAINTEXT"?

shroffk, 2019-08-30:
I can verify that I have 2 installations of the server talking to both local and remote kafka servers...haven't encountered any connection problems.

Here is an example of the nsls2 phoebus configuration
```
# Phoebus Alarm
org.phoebus.applications.alarm/server=xf23id1-srv3.cs.nsls2.local:9092
org.phoebus.applications.alarm/config_name=XF23ID_OPR
org.phoebus.applications.alarm/config_names=XF23ID_OPR
```

anantianshi, 2019-09-02:
Thank you very much,without "PLAINTEXT",it is ok.

closed 2019-12-09 by tanviash