ControlSystemStudio/phoebus #645: Alarm Notifier Behavior Question

tanviash, 2019-04-24:
@kasemir @shroffk 
I'm testing alarm notifier feature. Currently with my alarm tree containing just 1 PV sim://ramp which goes from OK-->MINOR-->MAJOR. I receive one notification for MINOR but there is no notification that comes for MAJOR that happened after MINOR. Is that expected behavior?

kasemir, 2019-04-25:
> Is that expected behavior?

Fundamentally, yes.

When the PV enters an alarm state, the notification is scheduled.
The notification usually has a rather long delay, like 15 minutes, because the idea is that notifications are a backup when local operators cannot acknowledge the alarm in time.
Once the scheduled time arrives and the alarm has not been acknowledged, the notification is generated with current information about the alarm.

For `sim:\\ramp(0, 10, 2)`, entering the minor alarm will schedule the notification. Assume a timeout of 20 seconds, it's sent 20 seconds later:
```
2019-04-25 10:41:57 INFO [org.phoebus.applications.alarm] /Accelerator/Main/sim:\/\/ramp(0, 10, 2): Schedule Mail in 20 s
2019-04-25 10:42:17 INFO [org.phoebus.applications.alarm] /Accelerator/Main/sim:\/\/ramp(0, 10, 2): Send email to [ky9@ornl.gov]
```

In the resulting email message, the alarm had latched up to MAJOR, while the current PV severity was OK, since that one keeps changing every 2 seconds:
```
Received: ..  25 Apr  2019 10:42:17 -0400
Subject: MAJOR alarm: sim://ramp(0, 10, 2)
/Accelerator/Main sim://ramp(0, 10, 2)
Description: sim://ramp(0, 10, 2)
Alarm Time: 2019-04-25 10:41:59.052
Alarm Severity: MAJOR, Status: HIHI, Value: 9.0
Current PV Severity: OK, Status: None
```

If the automated action delay is zero, the PV entering MINOR state will schedule and right away send the email:

```
2019-04-25 10:50:09 INFO [org.phoebus.applications.alarm] /Accelerator/Main/sim:\/\/ramp(0, 10, 2): Schedule Mail in 0 s
2019-04-25 10:50:09 INFO [org.phoebus.applications.alarm] /Accelerator/Main/sim:\/\/ramp(0, 10, 2): Send email to [ky9@ornl.gov]
```

The email will show the MINOR state:
```
/Accelerator/Main sim://ramp(0, 10, 2)
Description: sim://ramp(0, 10, 2)
Alarm Time: 2019-04-25 10:50:09.842
Alarm Severity: MINOR, Status: HIGH, Value: 8.0
Current PV Severity: MINOR, Status: HIGH
```

When the PV then 2 seconds later enters MAJOR state, there's no new notification because the alarm was already 'active', we did already send an email.
The point of the alarm system is to help, not annoy. An alarm that turns MINOR and then shortly thereafter MAJOR would fit the definition of a nuissance alarm.

tanviash, 2019-04-25:
@kay Thanks a lot for the explanation. Some follow up questions,

1. Right now with sim://ramp, moving from MINOR to MAJOR takes few seconds, what if for a real PV, moving from MINOR to MAJOR takes an hour, will then there be a MAJOR alarm notification after MINOR notification given that MINOR alarm has not be acknowledged and delay is set to 2 seconds?

2. Is the notifier behavior for BEAST same as Phoebus Alarm Server or there are some differences between the two?

3. If minor alarm happens after a MAJOR notification, I think that should not send any new notifications, correct?

4. When do we expect ACK notifications & OK (No active alarms) notifications?

kasemir, 2019-04-25:
The BEAST notifier and the way the phoebus alarm server handle automated actions are similar but different.

I'll try to find time to look at the details how a follow-up can be sent, i.e. once a MINOR message has been sent out, and then the PV goes into a higher severity, that would schedule another message.
Again in practice that's likely a bad scenario; a useful alarm does not send yet another notification.
Imagine operator receives MINOR alarm. Operator works on it. Now operator receives a MAJOR alarm? Operator is distracted by that, then annoyed because he's well aware of the issue and doesn't need another alarm for the same PV, so turns the alarm display off.
Still, I'll leave that for the person who creates the alarm PV to fix. The automated action will still trigger again.

As for cleared alarms, the behavior depends on `org.phoebus.applications.alarm/automated_action_followup`.
Here, we have that disabled because users didn't want all the "It's OK now" emails.

tanviash, 2019-04-25:
To summarize, this is how the current alarm notifier works (Please correct if I'm wrong). 
When PV goes into minor or major alarm for the first time, email notification is sent after passing of 'delay' seconds. Until this alarm is acknowledged and its value returns to OK, no other notification will be sent even if alarm goes into a more severe state or less severe state later.

kasemir, 2019-04-25:
Yes. Assume timeout of 10 seconds.

Minor alarm, acknowledged within 10 seconds => Nothing.

Minor alarm, not acknowledged within 10 seconds => Message about that minor alarm.

Minor alarm, turns into Major, not acknowledged after 10 seconds from Minor which started the timer => Message about that major alarm.

Once the message has been sent, and alarm is acknowledged, and automated_action_followup=mailto: ==> message about the new state, which might be minor_ack, major_ack, or OK.

After an alarm message has been sent, and alarm changes to a higher severity => No new message. THIS could be changed; could update the alarm server to send another message. In most cases, that's a result of bad alarm trigger configuration. The only useful scenario: Minor and Major alarms have very different reasons, are expected to be a long time apart.
For example:
a) Minor alarm on some medium pressure level. Guidance: Lower pressure by opening valve, turning that pump off etc.
b) Major alarm at some much higher pressure, could happen 2 hours after the minor alarm. Guidance: RUN! Nothing you can do.

In that case, both a Minor and Major alarm on the same PV would be valid.

tanviash, 2019-04-26:
@kasemir Thanks Kay for the explanation. I talked more to our operations people here and they think it is important for them to get notifications for more severe alarms that happens after minor notification. This is their way to find out from off-site if PV is trending towards more serious situation.

If you want, I can take a look at what needs to be changed in code or do you want to change it? Please let me know.

closed 2019-05-06 by tanviash