ControlSystemStudio/phoebus #647: alarm question

forestmojo, 2019-04-27:
hi, 

If I only want to use Phoebus to connect a alarm-server which was built by CSS, not using Kafka, how to write settings.ini file?
I checked the online help but can't find imformation.
Thanks a lot for your help.

kasemir, 2019-04-29:
The phoebus-based alarm UI and server use Kafka. The Eclipse-based alarm UI and server use JMS & RDB.  They both use the same XML format for importing/exporting the alarm configuration, but at runtime you use one or the other. The new UI cannot talk to the old server and vice versa.

forestmojo, 2019-04-30:
hi, Kay:

I follow the documention of phoebus about the alarm, when I try to run a alarm-server, I can't find service-alarm-server-0.0.1-SNAPSHOT.jar or similiar name jar file.
I have compiled the phoebus with maven. what's the reason? thanks a lot for your help.
  

Best wishes
Geyang 2019-04-30

###############

chenxin6, 2019-04-30:
I have not encountered the problem you mentioned. Can you describe in detail the steps you have installed?

forestmojo, 2019-04-30:
hi, 

When I try to rebuild the phoebus, it shows the errors like following, I am using Centos7, 
java version "12.0.1" 2019-04-16
Java(TM) SE Runtime Environment (build 12.0.1+12)
Java HotSpot(TM) 64-Bit Server VM (build 12.0.1+12, mixed mode, sharing)
what's the problem? thank you very much.
-----------------------------------------------------
FINE: Using URL examples:/monitors_textupdate.bob
[INFO] Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.004 s - in org.csstudio.display.builder.model.util.ModelResourceUtilTest
[INFO] Running org.csstudio.display.builder.model.StructuredWidgetPropertyUnitTest
Widget '' (plot) trace:
'pv_name' = ''
'foreground_color' = RGB(0,0,255)
<?xml version="1.0" encoding="UTF-8"?>
<display version="2.0.0">
  <widget type="plot" version="2.0.0">
    <trace>
      <pv_name>position</pv_name>
      <foreground_color>
        <color red="0" green="0" blue="255">
        </color>
      </foreground_color>
    </trace>
  </widget>
</display>

<?xml version="1.0" encoding="UTF-8"?>
<display version="2.0.0">
  <widget type="plot" version="2.0.0">
    <trace>
      <pv_name>position</pv_name>
      <foreground_color>
        <color red="255" green="255" blue="0">
        </color>
      </foreground_color>
    </trace>
  </widget>
</display>

DisplayModel ''
Widget '' (plot)
['type' = 'plot', 'name' = '', 'class' = DEFAULT, 'x' = 0, 'y' = 0, 'width' = 100, 'height' = 20, No action, 'rules' = [], 'scripts' = [], 'trace' = { 'pv_name' = 'position', 'foreground_color' = RGB(255,255,0) }]
'pv_name' = 'position'
['type' = 'plot', 'name' = '', 'class' = DEFAULT, 'x' = 0, 'y' = 0, 'width' = 100, 'height' = 20, No action, 'rules' = [], 'scripts' = [], 'trace' = { 'pv_name' = '', 'foreground_color' = RGB(0,0,255) }]
trace.pv_name: 'pv_name' = 'fred'
RGB(0,0,255)
[INFO] Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.017 s - in org.csstudio.display.builder.model.StructuredWidgetPropertyUnitTest
[INFO] Running org.csstudio.display.builder.PathTest
[INFO] Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in org.csstudio.display.builder.PathTest
[INFO] 
[INFO] Results:
[INFO] 
[ERROR] Failures: 
[ERROR]   PersistenceUnitTest.testWidgetReading:141 
Expected: "J?nner"
     but: was "J??nner"
[INFO] 
[ERROR] Tests run: 69, Failures: 1, Errors: 0, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for phoebus (parent) 0.0.1-SNAPSHOT:
[INFO] 
[INFO] phoebus (parent) ................................... SUCCESS [  6.220 s]
[INFO] dependencies ....................................... SUCCESS [  0.022 s]
[INFO] install-jars ....................................... SUCCESS [  0.687 s]
[INFO] phoebus-target ..................................... SUCCESS [  1.948 s]
[INFO] core ............................................... SUCCESS [  0.073 s]
[INFO] core-framework ..................................... SUCCESS [ 25.003 s]
[INFO] core-security ...................................... SUCCESS [  0.977 s]
[... 52 lines cut ...]
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  02:33 min
[INFO] Finished at: 2019-04-30T18:37:31+08:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.20:test (default-test) on project app-display-model: There are test failures.
[ERROR] 
[ERROR] Please refer to /root/phoebus/phoebus/app/display/model/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date]-jvmRun[N].dump, [date].dumpstream and [date]-jvmRun[N].dumpstream.
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :app-display-model
  

Best wishes
Geyang 2019-04-30

###############

kasemir, 2019-04-30:
Your build fails because a test related to character encoding, UTF-8 etc. fails.
For now ignore that and build without running the tests:
```
mvn -DskipTests clean install 
```

forestmojo, 2019-04-30:
it works. also if I export LANG="zh_CN.UTF-8", the compile can pass, but, new error comes:
[INFO] Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.009 s - in org.csstudio.javafx.rtplot.UnsignedsTest
[INFO] Running org.csstudio.javafx.rtplot.LinearTicksTest
Ticks for 1.0 .. 100.0:
2 4 6 8 '10' 12 14 16 18 '20' 22 24 26 28 '30' 32 34 36 38 '40' 42 44 46 48 '50' 52 54 56 58 '60' 62 64 66 68 '70' 72 74 76 78 '80' 82 84 86 88 '90' 92 94 96 98 '100' 
Ticks for -0.6 .. 0.6:
-0.6 -0.5 -0.5 -0.4 '-0.4' -0.4 -0.3 -0.3 -0.2 '-0.2' -0.2 -0.1 -0.1 -0.0 '0.0' 0.0 0.1 0.1 0.2 '0.2' 0.2 0.3 0.3 0.4 '0.4' 0.4 0.5 0.5 0.6 
Ticks for -100.0 .. 20000.0:
'0' 400 800 1200 1600 '2000' 2400 2800 3200 3600 '4000' 4400 4800 5200 5600 '6000' 6400 6800 7200 7600 '8000' 8400 8800 9200 9600 '10000' 10400 10800 11200 11600 '12000' 12400 12800 13200 13600 '14000' 14400 14800 15200 15600 '16000' 16400 16800 17200 17600 '18000' 18400 18800 19200 19600 '20000' 
Ticks for 10000.0 .. 1.0:
'10000' 9600 9200 8800 8400 '8000' 7600 7200 6800 6400 '6000' 5600 5200 4800 4400 '4000' 3600 3200 2800 2400 '2000' 1600 1200 800 400 
[ERROR] Tests run: 4, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 0.017 s <<< FAILURE! - in org.csstudio.javafx.rtplot.LinearTicksTest
[ERROR] testNormalTicks(org.csstudio.javafx.rtplot.LinearTicksTest)  Time elapsed: 0.014 s  <<< FAILURE!
java.lang.AssertionError: 

Expected: "'0' 1000 2000 3000 4000 '5000' 6000 7000 8000 9000 '10000' 11000 12000 13000 14000 '15000' 16000 17000 18000 19000 '20000' "
     but: was "'0' 400 800 1200 1600 '2000' 2400 2800 3200 3600 '4000' 4400 4800 5200 5600 '6000' 6400 6800 7200 7600 '8000' 8400 8800 9200 9600 '10000' 10400 10800 11200 11600 '12000' 12400 12800 13200 13600 '14000' 14400 14800 15200 15600 '16000' 16400 16800 17200 17600 '18000' 18400 18800 19200 19600 '20000' "
	at org.csstudio.javafx.rtplot.LinearTicksTest.testNormalTicks(LinearTicksTest.java:73)

[INFO] 
[INFO] Results:
[INFO] 
[ERROR] Failures: 
[ERROR]   LinearTicksTest.testNormalTicks:73 
Expected: "'0' 1000 2000 3000 4000 '5000' 6000 7000 8000 9000 '10000' 11000 12000 13000 14000 '15000' 16000 17000 18000 19000 '20000' "
     but: was "'0' 400 800 1200 1600 '2000' 2400 2800 3200 3600 '4000' 4400 4800 5200 5600 '6000' 6400 6800 7200 7600 '8000' 8400 8800 9200 9600 '10000' 10400 10800 11200 11600 '12000' 12400 12800 13200 13600 '14000' 14400 14800 15200 15600 '16000' 16400 16800 17200 17600 '18000' 18400 18800 19200 19600 '20000' "
[INFO] 
[ERROR] Tests run: 21, Failures: 1, Errors: 0, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for phoebus (parent) 0.0.1-SNAPSHOT:
[INFO] 
[INFO] phoebus (parent) ................................... SUCCESS [  0.665 s]
[INFO] dependencies ....................................... SUCCESS [  0.033 s]
[INFO] install-jars ....................................... SUCCESS [  0.937 s]
[INFO] phoebus-target ..................................... SUCCESS [  1.773 s]
[INFO] core ............................................... SUCCESS [  0.016 s]
[INFO] core-framework ..................................... SUCCESS [ 23.405 s]
[INFO] core-security ...................................... SUCCESS [  1.021 s]
[INFO] core-util .......................................... SUCCESS [  1.500 s]
[INFO] core-pv ............................................ SUCCESS [ 46.003 s]
[INFO] core-ui ............................................ SUCCESS [  6.312 s]
[INFO] core-logbook ....................................... SUCCESS [  0.261 s]
[INFO] core-types ......................................... SUCCESS [  0.182 s]
[INFO] core-formula ....................................... SUCCESS [  0.895 s]
[INFO] core-email ......................................... SUCCESS [  0.073 s]
[INFO] app ................................................ SUCCESS [  0.015 s]
[INFO] app-diag ........................................... SUCCESS [  0.211 s]
[INFO] app-log-configuration .............................. SUCCESS [  0.195 s]
[INFO] app-probe .......................................... SUCCESS [  0.282 s]
[INFO] app-email .......................................... SUCCESS [  0.316 s]
[INFO] app-logbook ........................................ SUCCESS [  0.016 s]
[INFO] app-logbook-utility ................................ SUCCESS [  0.121 s]
[INFO] app-logbook-ui ..................................... SUCCESS [  1.751 s]
[INFO] app-pvtable ........................................ SUCCESS [  0.438 s]
[INFO] app-pvtree ......................................... SUCCESS [  1.141 s]
[INFO] app-update ......................................... SUCCESS [  0.243 s]
[INFO] app-filebrowser .................................... SUCCESS [  0.310 s]
[INFO] app-logbook-inmemory ............................... SUCCESS [  0.140 s]
[INFO] app-logbook-olog ................................... SUCCESS [  0.323 s]
[INFO] app-rtplot ......................................... FAILURE [  6.861 s]
[INFO] app-databrowser .................................... SKIPPED
[INFO] app-display ........................................ SKIPPED
[INFO] app-display-model .................................. SKIPPED
[INFO] app-display-representation ......................... SKIPPED
[INFO] app-3dViewer ....................................... SKIPPED
[INFO] app-display-representation-javafx .................. SKIPPED
[INFO] app-display-runtime ................................ SKIPPED
[... cut ...]
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :app-rtplot

forestmojo, 2019-04-30:
all the problems are solved.

closed 2019-04-30 by forestmojo

kasemir, 2019-04-30:
>   if I export LANG="zh_CN.UTF-8", the compile can pass, but

... different test errors..
I can confirm that the `org.csstudio.javafx.rtplot.LinearTicksTest` fails in that locale.
Not all tests are prepared to work the same way in all locales.
To run tests, try setting environment variable `LANG` to `en_US.UTF-8`.