ControlSystemStudio/phoebus #658: import phoebus in eclipse

forestmojo, 2019-05-05:
hi, 

I followed the instruction:
1.install Eclipse Oxygen 4.7.1a, jdk9.04
2.In Help/Eclipse Marketplace, search for Maven Integration for Eclipse Luna or newer
3.Use File/Import/Maven/Existing Maven Projects to import the phoebus source code.
4.change JRE System Library from J2SE-1.5 to default jdk-9.04
but there are many erros:
Description	Resource	Path	Location	Type
'<>' cannot be used with anonymous classes	FileHelper.java	/core-framework/src/main/java/org/phoebus/framework/workbench	line 33	Java Problem
'<>' operator is not allowed for source level below 1.7	AdapterService.java	/core-framework/src/main/java/org/phoebus/framework/adapter	line 35	Java Problem
'<>' operator is not allowed for source level below 1.7	AdapterService.java	/core-framework/src/main/java/org/phoebus/framework/adapter	line 41	Java Problem
'<>' operator is not allowed for source level below 1.7	ApplicationService.java	/core-framework/src/main/java/org/phoebus/framework/workbench	line 36	Java Problem
'<>' operator is not allowed for source level below 1.7	ApplicationService.java	/core-framework/src/main/java/org/phoebus/framework/workbench	line 39	Java Problem
'<>' operator is not allowed for source level below 1.7	CommandExecutor.java	/core-framework/src/main/java/org/phoebus/framework/jobs	line 65	Java Problem
'<>' operator is not allowed for source level below 1.7	FileHelper.java	/core-framework/src/main/java/org/phoebus/framework/workbench	line 33	Java Problem
'<>' operator is not allowed for source level below 1.7	History.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 29	Java Problem
'<>' operator is not allowed for source level below 1.7	History.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 65	Java Problem
'<>' operator is not allowed for source level below 1.7	JobManager.java	/core-framework/src/main/java/org/phoebus/framework/jobs	line 34	Java Problem
'<>' operator is not allowed for source level below 1.7	JobManager.java	/core-framework/src/main/java/org/phoebus/framework/jobs	line 83	Java Problem
'<>' operator is not allowed for source level below 1.7	LocProposal.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 66	Java Problem
'<>' operator is not allowed for source level below 1.7	LocProposal.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 152	Java Problem
'<>' operator is not allowed for source level below 1.7	LocProposalProvider.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 48	Java Problem
'<>' operator is not allowed for source level below 1.7	Macros.java	/core-framework/src/main/java/org/phoebus/framework/macros	line 37	Java Problem
'<>' operator is not allowed for source level below 1.7	Macros.java	/core-framework/src/main/java/org/phoebus/framework/macros	line 242	Java Problem
'<>' operator is not allowed for source level below 1.7	MqttProposal.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 49	Java Problem
'<>' operator is not allowed for source level below 1.7	MqttProposal.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 82	Java Problem
'<>' operator is not allowed for source level below 1.7	MqttProposalProvider.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 48	Java Problem
'<>' operator is not allowed for source level below 1.7	NLS.java	/core-framework/src/main/java/org/phoebus/framework/nls	line 70	Java Problem
'<>' operator is not allowed for source level below 1.7	PreferencesReader.java	/core-framework/src/main/java/org/phoebus/framework/preferences	line 100	Java Problem
'<>' operator is not allowed for source level below 1.7	Proposal.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 63	Java Problem
'<>' operator is not allowed for source level below 1.7	ProposalService.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 49	Java Problem
'<>' operator is not allowed for source level below 1.7	ProposalService.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 53	Java Problem
'<>' operator is not allowed for source level below 1.7	PvaProposalProvider.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 46	Java Problem
'<>' operator is not allowed for source level below 1.7	RDBConnectionPool.java	/core-framework/src/main/java/org/phoebus/framework/rdb	line 46	Java Problem
'<>' operator is not allowed for source level below 1.7	RDBConnectionPool.java	/core-framework/src/main/java/org/phoebus/framework/rdb	line 48	Java Problem
'<>' operator is not allowed for source level below 1.7	SelectionService.java	/core-framework/src/main/java/org/phoebus/framework/selection	line 16	Java Problem
'<>' operator is not allowed for source level below 1.7	SelectionService.java	/core-framework/src/main/java/org/phoebus/framework/selection	line 18	Java Problem
'<>' operator is not allowed for source level below 1.7	SimProposal.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 72	Java Problem
'<>' operator is not allowed for source level below 1.7	SimProposal.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 138	Java Problem
'<>' operator is not allowed for source level below 1.7	SimProposalProvider.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 57	Java Problem
'<>' operator is not allowed for source level below 1.7	SimProposalProvider.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 111	Java Problem
'<>' operator is not allowed for source level below 1.7	XMLMementoTree.java	/core-framework/src/main/java/org/phoebus/framework/persistence	line 186	Java Problem
Cannot refer to the non-final local variable adapterFactory defined in an enclosing scope	AdapterService.java	/core-framework/src/main/java/org/phoebus/framework/adapter	line 37	Java Problem
Cannot refer to the non-final local variable provider defined in an enclosing scope	ProposalService.java	/core-framework/src/main/java/org/phoebus/framework/autocomplete	line 86	Java Problem
Cannot refer to the non-final local variable selection defined in an enclosing scope	SelectionUtil.java	/core-framework/src/main/java/org/phoebus/framework/selection	line 26	Java Problem
Constructor references are allowed only at source level 1.8 or above	ResourceParser.java	/core-framework/src/main/java/org/phoebus/framework/util	line 190	Java Problem
Default methods are allowed only at source level 1.8 or above	AppDescriptor.java	/core-framework/src/main/java/org/phoebus/framework/spi	line 31	Java Problem
Default methods are allowed only at source level 1.8 or above	AppDescriptor.java	/core-framework/src/main/java/org/phoebus/framework/spi	line 40	Java Problem
Default methods are allowed only at source level 1.8 or above	AppDescriptor.java	/core-framework/src/main/java/org/phoebus/framework/spi	line 48	Java Problem
Default methods are allowed only at source level 1.8 or above	AppDescriptor.java	/core-framework/src/main/java/org/phoebus/framework/spi	line 61	Java Problem
Default methods are allowed only at source level 1.8 or above	AppInstance.java	/core-framework/src/main/java/org/phoebus/framework/spi	line 13	Java Problem
Default methods are allowed only at source level 1.8 or above	AppInstance.java	/core-framework/src/main/java/org/phoebus/framework/spi	line 27	Java Problem
Default methods are allowed only at source level 1.8 or above	AppInstance.java	/core-framework/src/main/java/org/phoebus/framework/spi	line 38	Java Problem
Default methods are allowed only at source level 1.8 or above	AppResourceDescriptor.java	/core-framework/src/main/java/org/phoebus/framework/spi	line 26	Java Problem
Default methods are allowed only at source level 1.8 or above	AppResourceDescriptor.java	/core-framework/src/main/java/org/phoebus/framework/spi	line 50	Java Problem
Lambda expressions are allowed only at source level 1.8 or above	AdapterService.java	/core-framework/src/main/java/org/phoebus/framework/adapter	line 31	Java Problem
Lambda expressions are allowed only at source level 1.8 or above	AdapterService.java	/core-framework/src/main/java/org/phoebus/framework/adapter	line 33	Java Problem
Lambda expressions are allowed only at source level 1.8 or above	ExternalApplication.java	/core-framework/src/main/java/org/phoebus/framework/workbench	line 69	Java Problem
[... cut ...]
The project was not built since it depends on phoebus-target, which has build path errors	service-alarm-logger		Unknown	Java Problem
The project was not built since it depends on phoebus-target, which has build path errors	service-alarm-server		Unknown	Java Problem
The project cannot be built until its prerequisite core-util is built. Cleaning and building all projects is recommended	service-archive-engine		Unknown	Java Problem
The project was not built since it depends on phoebus-target, which has build path errors	service-scan-server		Unknown	Java Problem

kasemir, 2019-05-05:
> I followed the instruction:
> ... source level below 1.7 

Missed a step:

    Check Eclipse Preferences:
    ...
    Java, Compiler: JDK Compliance should be "9" or higher

forestmojo, 2019-05-05:
hi, Kay:

I think I have modified as it's mentioned in:
Check Eclipse Preferences:
Java, Installed JREs: JDK 9-or-later should be the default
Java, Compiler: JDK Compliance should be "9" or higher

kasemir, 2019-05-05:
Check the settings of affected projects like `core-framework`: `..operator .. not allowed for source level below 1.7`. The error indicates that the project is configured for source level 1.7 instead of 9 or higher, hence the error.

forestmojo, 2019-05-06:
Now it seems the errors are related with library dependencies:
Description	Resource	Path	Location	Type
Plugin execution not covered by lifecycle configuration: org.apache.maven.plugins:maven-install-plugin:2.5.2:install-file (execution: install-pbraw, phase: validate)	pom.xml	/install-jars	line 68	Maven Project Build Lifecycle Mapping Problem
Project 'app-display-representation-javafx' is missing required Java project: 'app-3d-viewer'	app-display-representation-javafx		Build path	Build Path Problem
Project 'app-logbook-ui' is missing required library: '/dependencies/phoebus-target/target/lib/assertj-core-3.9.1.jar'	app-logbook-ui		Build path	Build Path Problem
Project 'app-logbook-ui' is missing required library: '/dependencies/phoebus-target/target/lib/testfx-core-4.0.13-alpha.jar'	app-logbook-ui		Build path	Build Path Problem
Project 'app-logbook-ui' is missing required library: '/dependencies/phoebus-target/target/lib/testfx-junit-4.0.13-alpha.jar'	app-logbook-ui		Build path	Build Path Problem
Project 'core-pv' is missing required library: '/phoebus-target/target/lib/epics-util-1.0.1-SNAPSHOT.jar'	core-pv		Build path	Build Path Problem
Project 'core-pv' is missing required library: '/phoebus-target/target/lib/org.eclipse.paho.client.mqttv3-1.2.0.jar'	core-pv		Build path	Build Path Problem
Project 'core-pv' is missing required library: '/phoebus-target/target/lib/vtype-1.0.1-SNAPSHOT.jar'	core-pv		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/activation-1.1.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/commonj.sdo-2.1.1.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/commons-codec-1.10.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/commons-logging-1.1.3.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/controlsfx-9.0.0.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/derby-10.14.1.0.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/eclipselink-2.7.0.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/elasticsearch-6.3.1.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/elasticsearch-cli-6.3.1.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/elasticsearch-core-6.3.1.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/elasticsearch-rest-client-6.3.1.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/elasticsearch-rest-high-level-client-6.3.1.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/elasticsearch-secure-sm-6.3.1.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/elasticsearch-x-content-6.3.1.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/epics-pvaccess-5.1.3-SNAPSHOT.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/epics-pvdata-6.1.3-SNAPSHOT.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/epics-util-1.0.1-SNAPSHOT.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/freetts-1.2.2.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/guava-21.0.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/hamcrest-all-1.3.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/hamcrest-core-1.3.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/HdrHistogram-2.1.9.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/hibernate-validator-6.0.12.Final.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/hppc-0.7.1.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/httpasyncclient-4.1.2.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/httpclient-4.5.2.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/httpcore-4.4.5.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/httpcore-nio-4.4.5.jar'	phoebus-target		Build path	Build Path Problem
Project 'phoebus-target' is missing required library: 'target/lib/jackson-annotations-2.9.4.jar'	phoebus-target		Build path	Build Path Problem
[... cut ...]
The project was not built since it depends on phoebus-target, which has build path errors	service-alarm-logger		Unknown	Java Problem
The project was not built since it depends on phoebus-target, which has build path errors	service-alarm-server		Unknown	Java Problem
The project cannot be built until its prerequisite core-util is built. Cleaning and building all projects is recommended	service-archive-engine		Unknown	Java Problem
The project was not built since it depends on phoebus-target, which has build path errors	service-scan-server		Unknown	Java Problem

kasemir, 2019-05-06:
`Project 'phoebus-target' is missing required library: 'target/lib/....jar'`

See Target Platform:

All external dependencies are expected in dependencies/target/lib. They could be obtained by expanding a zip-ed phoebus target from an existing build setup, or via one initial maven run:
```
mvn clean verify -f dependencies/pom.xml
```

forestmojo, 2019-05-06:
seems the dawn if nearby, the errors are like following:
Description	Resource	Path	Location	Type
Plugin execution not covered by lifecycle configuration: org.apache.maven.plugins:maven-install-plugin:2.5.2:install-file (execution: install-pbraw, phase: validate)	pom.xml	/install-jars	line 68	Maven Project Build Lifecycle Mapping Problem
Project 'app-display-representation-javafx' is missing required Java project: 'app-3d-viewer'	app-display-representation-javafx		Build path	Build Path Problem
Project 'product' is missing required Java project: 'app-3d-viewer'	product		Build path	Build Path Problem
The project cannot be built until build path errors are resolved	app-display-representation-javafx		Unknown	Java Problem
The project cannot be built until build path errors are resolved	product		Unknown	Java Problem
The project was not built since it depends on app-display-representation-javafx, which has build path errors	app-display-editor		Unknown	Java Problem
The project was not built since it depends on app-display-representation-javafx, which has build path errors	app-display-runtime		Unknown	Java Problem
Type mismatch: cannot convert from ObservableList<Callback<AlarmInfoRow,Observable[]>> to ObservableList<AlarmInfoRow>	AlarmTableUI.java	/app-alarm-ui/src/main/java/org/phoebus/applications/alarm/ui/table	line 94	Java Problem
Type mismatch: cannot convert from ObservableList<Callback<AlarmInfoRow,Observable[]>> to ObservableList<AlarmInfoRow>	AlarmTableUI.java	/app-alarm-ui/src/main/java/org/phoebus/applications/alarm/ui/table	line 95	Java Problem
Type mismatch: cannot convert from ObservableList<Callback<TableItemProxy,Observable[]>> to ObservableList<TableItemProxy>	PVTable.java	/app-pvtable/src/main/java/org/phoebus/applications/pvtable/ui	line 108	Java Problem

closed 2019-05-08 by forestmojo