ControlSystemStudio/phoebus #660: Is there any plan to integrate olog server into phoebus?

yifans, 2019-05-07:
Is there any plan to integrate olog server (https://github.com/Olog/olog-service) into phoebus as a server in the future, like the alarm-server and scan-server?

shroffk, 2019-05-07:
yes,
I am working on a springboot based olog with elasticsearch as its back end. The client for this will be in phoebus...we might even consider moving the server code under services.
https://github.com/shroffk/olog-es

forestmojo, 2019-05-08:
Wonderful. How about webopi? does phoebus support this now? I didn't see it in doc file.

shroffk, 2019-05-21:
This is not sure yet...while we are including sprinboot service under phoebus/services the olog service pulls in a lot of dependencies.

shroffk, 2021-03-09:
https://github.com/Olog/olog-es

closed 2021-03-09 by shroffk