ControlSystemStudio/phoebus #668: Display Builder Editor: widget moving misbehavior when zoomed in

pavel-ch, 2019-05-16:
In the situation when:
- The editor pane is zoomed in (zoom drop-down box or Ctrl-mouse wheel)
- The editor pane is scrolled to non-zero scroll position horizontally and/or vertically

When trying to move or resize widgets, comes to unexpected jumps in position or size.

The reason not found (resp. not looked for) so far.

claudio-rosati, 2019-05-16:
I've seen the problem too and it is part of my issues to solve. My impression is that some operations (mostly related to D&D) doesn't consider the current zoom level in their computations.

kasemir, 2019-06-19:
The bug is in the AutoScrollHandler.
No more 'jumps' when disabling its `scroll` method.

kasemir, 2019-06-19:
I was wrong. Had coincidentally disabled the "Snap to widgets" alignment, which is the actual issue.
The baby-blue alignment lines which "Snap to widgets" adds to the display cause the confusion.
With "Snap to widgets" disabled, you can drag/resize widgets in a zoomed editor.

kasemir, 2019-06-20:
Fixed

closed 2019-06-20 by kasemir