ControlSystemStudio/phoebus #671: Display Builder Editor: EditEmbeddedDisplayAction: decoupling from EmbeddedDisplayWidget

pavel-ch, 2019-05-17:
In my product, I have a couple of widgets using embedded displays in a slightly different way from the `EmbeddedDisplayWidget` (an example is a symbol widget using Groups, working internally a bit more effective than changing group via a rule in `EmbeddedDisplayWidget`). Would be nice to have the `EditEmbeddedDisplayAction` activated in the context menu in Display Editor also for these widgets.
This requires to decouple the `EditEmbeddedDisplayAction` from `EmbeddedDisplayWidget`.
Prototype follows in PR, in principle instead of using `instanceof EmbeddedDisplayWidget` check, I check for propFile property and the `bob` extension inside.

kasemir, 2019-05-17:
You're providing many useful little fixes, thanks!
Are your widget types useful as a general addition?
Care to join our monthly online meetings to show them there?
Are you by chance attending the upcoming EPICS meeting, https://epics.anl.gov/meetings.php?

pavel-ch, 2019-05-20:
You are welcome!
I am customizing the functions at some level to our environment (we are active in electric substation automation). I am in the slow start phase of proof-of concept with Phoebus integration atm. We are using the MQTT customized driver to our server(s) with JSON attribute transport. As our attributes set differs from CA, I had to use customized widgets, representations and runtimes along with tooltips, context menus etc.
This means, unfortunately, the widgets are not useful for the general product.
So at least I am trying to contribute with generally useful things like fixes and/or some ideas.
Not being part of your awesome cooperation on accelerators control, I think that my joining monthly meetings is of no added value for you. And unfortunately we cannot meet on EPICS meetings :-(
Anyway I hope all I am doing is in conformance with Phoebus licensing!

closed 2019-05-20 by pavel-ch

kasemir, 2019-05-20:
Very glad to hear that the MQTT support which was mostly added as a proof-of-concept for supporting additional network protocols is being used!
I'll try to contact you via email for some more info..