ControlSystemStudio/phoebus #675: alarm server configuration import failure

shroffk, 2019-05-21:
The alarm server fails to correctly import the alarm configuration file exported from the beast server.
1. disabled alarms are not loaded at all

2. the latch, annunciation, delay parameters are ignored and replaced with default
[image]

[image]

kasemir, 2019-05-21:
Can you attach an example file that you'd like to import?

shroffk, 2019-05-21:
I tried but github is not a fan of xml...sorry but I only saw zip in the support list

[fixed_alarm.zip](https://github.com/shroffk/phoebus/files/3203238/fixed_alarm.zip)

kasemir, 2019-05-21:
When I create the topics for XF23ID_OPR, I get
```
WARNING: Due to limitations in metric names, topics with a period ('.') or underscore ('_') could collide. To avoid issues it is best to use either, but not both.
Created topic "XF23ID_OPR".
```

shroffk, 2019-05-21:
We don't use "." with any other topic only "_" is used...and currently the kafka server has only topics for the config "XF23ID_OPR"

kasemir, 2019-05-21:
>  disabled alarms are not loaded at all

Loads for me.

XML:
```
<pv name="XF:23ID1-CT{IOC:SRV1}:MEM_USED_P">
  <description>XF23ID1-SRV1 System Memory</description>
  <enabled>false</enabled>
  <latching>true</latching>
</pv>
```

Kafka:
```
CreateTime:1558452023028: /XF23ID_OPR/23-ID/Control System/Computer Systems/XF:23ID1-CT{IOC:SRV1}:MEM_USED_P: {"user":"ky9","host":"diane.ornl.gov","description":"XF23ID1-SRV1 System Memory","enabled":false}
```

UI:
[image]

>  the latch, annunciation, delay parameters are ignored and replaced with default

latching is by default set. So `<latching>true</latching>` doesn't change anything.
<latching>false</latching>` would, but that's not in the XML file.

Delay gets imported when I check this one in the UI:
```
<pv name="XF:23ID1-CT{IOC:IOC1}:SYS_CPU_LOAD">
    <description>XF23ID1-IOC1 System CPU</description>
   <latching>true</latching>
   <delay>60.0</delay>
    <count>0</count>
</pv>
```
[image]

Do you have errors in the import?

Smaller example with entries that don't import for you?

kasemir, 2019-05-21:
> latching is by default set.

Looks like that was different in the old import/export tool:
Latching and annunciating were written as 'true', they defaulted to 'false'.
Never noticed because all our alarms are latching and annunciated..

I'll change it to be compatible to the previous version:
Latching or annunciating not in the XML -> default to false.

kasemir, 2019-05-21:
Here's the delay that I get for `XF:23ID-CT{RG:D2}T:2-I` where your import has 0 seconds:

Both the kafka dump and the UI have delay: 30 seconds

```
CreateTime:1558452023033: /XF23ID_OPR/23-ID/Control System/Rack Temp Monitoring/XF:23ID-CT{RG:D2}T:2-I: {"user":"ky9","host":"diane.ornl.gov","description":"XF:23ID-CT{RG:D2}T:2-I","delay":30}
```

[image]

shroffk, 2019-05-21:
Here is an example configuration file.

To summarize our findings from the hangout:

Loading the file for the first time into an empty configuration works fine.
Reload the file again which requires deleted and then re adding the nodes/pvs causes some discrepancy.

[export_fixed_alarm.zip](https://github.com/shroffk/phoebus/files/3203764/export_fixed_alarm.zip)

kasemir, 2019-05-21:
The import sent two kafka messages for each item: One message with the item path and some default content, then another message right away with the actual content.
The above commit reduces it to only one message.

But either way, I can import that config many times and the client looks OK, so issue doesn't reproduce here, just like you said it's fine in your office setup.
In the failing setup, when running with logging set to 
`org.phoebus.applications.alarm.level = ALL`,
does it echo all the Kafka messages, like
```
.. [org.phoebus.applications.alarm]
 XF23ID_OPR @ 2019-05-21 14:42:18.062
 /XF23ID_OPR/Computer Systems/XF:23ID1-CT{IOC:SRV1}:MEM_USED_P
 =
 {..."description":"XF23ID1-SRV1 System Memory",
  "enabled":false,
 "annunciating":false}
```
?

shroffk, 2019-05-23:
Ok, I started with a fresh alarm setup.

I was able to reproduce the unreliable behaviour when loading/reloading the above alarm configuration.

[image]

however, this time I did get some error messages in the alarm server.

```
XF23ID_OPR > 2019-05-23 11:58:26 SEVERE [org.phoebus.applications.alarm] Error in delayed alarm update
java.lang.NullPointerException
	at org.phoebus.applications.alarm.server.AlarmServerPV$1.alarmStateChanged(AlarmServerPV.java:113)
	at org.phoebus.applications.alarm.server.AlarmLogic.updateState(AlarmLogic.java:400)
	at org.phoebus.applications.alarm.server.AlarmLogic.delayedStateUpdate(AlarmLogic.java:414)
	at org.phoebus.applications.alarm.server.DelayedAlarmUpdate.lambda$schedule_update$1(DelayedAlarmUpdate.java:101)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:514)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1135)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:844)
```

kasemir, 2019-05-23:
The alarm server NullPointerException is likely unrelated, a result of some PV getting an update while the configuration is deleted and the PV's parent entry in the model is gone. Not good, but unrelated.

Concentrating just on the alarm tree as a client and the import, can you now reproduce it?
Or does it only happen when the alarm server is also running at the same time??

When you configure logging as
```
org.phoebus.applications.alarm.level = ALL
```
the UI should log all received Kafka messages in `AlarmClient.checkUpdates`.
Does it get all the messages, same shown by `./monitor_topic.sh XF23ID_OPR`?

If I could reproduce it, I would set a breakpoint in `AlarmClient.checkUpdates` and trace what's happening. Maybe a breakpoint conditional on `node_config.contains("\"enabled\":false")` to see specifically what's happening with the disabled entry.

kasemir, 2019-05-23:
Also, in the problematic state, when closing the alarm tree and opening it again, will it show the correct model? I'm unclear if the client is seeing something different, or if the tree code mishandles the updates.

kasemir, 2019-05-23:
OK, running the Alarm Server I now also see something odd.

Run alarm tree, import config -> Works every time.
Run alarm server, run alarm tree, import config -> Sometimes odd.

When closing the alarm tree and opening it again, or selecting a different alarm config name, the alarm tree is again fine.
The issue seems to be in the order of messages.
When the alarm server runs and sees that an item is removed, it sends a state update with null.
The client sees the config update and the state update is unpredictable order.
Looking deeper and fixing this might have to wait until after the EPICS meeting.

kasemir, 2019-05-23:
Symptom of the issue:  Run alarm server, run alarm tree with logging, import config, and the end of the alarm tree log might show

```
XF23ID_OPRState @ 2019-05-23 15:08:55.354 /XF23ID_OPR/Computer Systems = null
Detach /XF23ID_OPR/Computer Systems from parent
```

So after it first handled the config update OK, it then receives the late alarm server state update that resulted from deleting the config.

shroffk, 2019-05-24:
Ok...
I am glad we have a reproduce able problem.
I have managed to get a virtual box image where this problem can be reliably produced...

Maybe if there is time we can add this to the agenda of the developers meeting,

shroffk, 2019-06-06:
The AlarmMessage is to test the use of object mapper for encoding and decoding state and config messages being published to Kafka... we might be able to simplify this even more.

kasemir, 2019-06-12:
There is an ordering issue which results when loading/reloading the configuration while the alarm server is running.

The configuration and state updates use different topics.
When entries are deleted, _both_ will include `null` messages to remove those items.
While messages are ordered within a topic, messages from different topics are received at unpredictable times.

We tried to work around this by tracking the time of the last state update etc.
Better fix is to simply place both config and state messages in the main topic.
Extend the message `key` to indicate the message type, because otherwise a config and state message for the same path would be compacted, deleting either the config or the state.

PR #687 (2019-06-24): Alarm Server & UI: Merge config and state topic
Fixes #675
closed 2019-06-24 by shroffk