ControlSystemStudio/phoebus #676: alarm server kafka configuration topic management

shroffk, 2019-05-21:
Kafka alarm topics were created on a single node installation of kafka using the create_alarm_topics.sh

The kafka server was rebooted as per the instructions to ensure the proper behaviour of the "compact" delete mechanism for the config topic.

While most installations normally work well after the above actions...on some occasions discrepancies are observed between the config topic as seen by the alarm server and as seen by the client

e.g.

[image]
The alarm server sees an alarm configuration with only 2 nodes.

[image]
The alarm client at the same time sees 4 nodes (older)

While this is surely a problem with the kafka configuration I have not been able to reliable reproduce this configuration mismatch state. Once this situation arises, a re importing of the alarm configuration resolves the issue.

kasemir, 2019-05-21:
In that situation, what's the output of `monitor_topic.sh XF23ID_OPR`?
Are there eventually null messages for those older nodes to delete them?

shroffk, 2019-05-21:
Currently I have re imported the configuration and this problem is no longer there.
I have set up the logger and I will search the elastic log to see if the null messages were indeed sent out.

kasemir, 2019-06-12:
I suspect the underlying issue is the same as #675. Will describe there.

kasemir, 2019-08-12:
Closing as duplicate of 675

closed 2019-08-12 by kasemir