ControlSystemStudio/phoebus #679: Display Editor - enhancements

pavel-ch, 2019-05-23:
Working on the demo screen, some hints from the work with Display Editor on Windows OS:
1. PolyLine Editor: line points placing should follow grid constraints when the Grid Alignment is On
2. Group behavior
    a) It is necessary to be able to start "lasso" selection gesture from inside the group. Currently it is very impractical in the case of heavy Groups usage - having to find a piece of free display to start lasso.
    b) Possibility to add a new widget to the Group from the Widgets panel - currently not possible
3. Macroized Property: 
    a) especially for the PV Name, it would be nice to add some kind of interactive support for "macroizing" the string using current macros ("reverse" macro application)
    b) enhance the PV completion, allowing work with macroized PV Names, if possible
4. Better SAVE function availability:
    a) sometimes Ctrl-S does not work from unknown reason (perhaps focus problem), also the context menu must be clicked twice before it appears
    b) Place SAVE button to Editor toolbar and/or Editor display context menu would help

Thank you for your opinions.

kasemir, 2019-05-23:
All good. Specifically, I can confirm the Save-As issue, have reopened associated ticket. Would be very helpful to get some reproducible recipe, since I can get into the situation where save-as is disabled, but then can't remember how I got there.
I won't get to do anything about any of these in the next 2 weeks, though.

pavel-ch, 2019-05-23:
Fine. Meanwhile, I can look into items and send usable results as PRs (if available)

DorianHas, 2019-08-12:
I would add to this list the ability to rotate widgets. This could be very useful when modelizing beam line.

It could be associated to Ctrl-R or changed in Position section in  widget properties.

kasemir, 2019-08-29:
Re Group behavior, starting a 'rubberband' (lasso) selection from inside the group, that's now possible by holding the Alt key. Ordinarily, trying to 'rubberband' within a group doesn't work because the initial click selects the group. Now, holding Alt, the initial click is not selecting the group but instead allows starting a rubberband selection of items within the group.

kasemir, 2019-08-29:
Re "add a new widget to the Group from the Widgets panel - currently not possible":
I tried dragging a new widget from the list of widgets into a group. Works fine. As widget is dragged out of the list of widgets and over a group, the group is highlighted. When then letting go, the widget is added to that group.

kasemir, 2019-08-29:
Re Group behavior, starting a 'rubberband' (lasso) selection from inside the group, 
that's now also possible for the Tabs widget:
While holding Alt, you can rubberband-select items within the currently active tab.

kasemir, 2019-08-30:
Re PolyLine Editor:.. grid constraints ..

has been implemented

pavel-ch, 2019-09-03:
Re "add a new widget to the Group from the Widgets panel - currently not possible":

Sorry for me not being able to contribute due to the full load of another tasks.
Shortly:
Drop works both outside/inside group. Click and drag (pressing left mouse and move) works outside a group, inside not.
Bug: the drop gesture places the new widget into a false position when zoomed.

Thanks for all the work.

kasemir, 2019-09-03:
OK, got it. We're down to _drop_ having a problem when dropping inside a group while zoomed. Applies to both dropping a new widget that was dragged out of the widget palette, or when using Ctrl-Drag to copy-drag an existing widget.

kasemir, 2019-09-03:
Starting to look into this, it actually seems unrelated to groups or zoom levels.
It's about the origin of the viewport.
As long as the editor is looking at the top-left corner of the display, you can be zoomed, and dropping widgets inside or outside of groups works fine.

But when you pan the display to a viewport other than the top-left corner, simply dropping a widget has it appear at a location that's wrong, presumably related to the viewport offset.
When dealing with a group and zooming in, it's quite likely that you also end up with a viewport offset, so that explains why we first saw the issue in that scenario.

pavel-ch, 2019-09-03:
Yes right, this is for _drop_ issue. Seems you are on track, issue is really related to the viewport and not to zoom.

The 

> Click and drag (pressing left mouse and move) works outside a group, inside not

 is another (MINOR) issue, reproducible by the following:
Click (=left mouse down and up) on the LED button in Widget Pane. The button is "pressed".
Press left mouse button on a free place on display and move mouse with button pressed. Release button.
You get a new LED inside the rectangle you "drawn" with the mouse.
Doing the same on a group does nothing.

kasemir, 2019-09-03:
Regarding the minor issue:
You can now
1) Select a new widget to be added by clicking it in the widget palette
2) Hold Alt while rubberbanding the desired location and size inside a group to create a new widget there

Without the Alt, you're right, clicking in the group will select that group. But with the new Alt behavior, selection is suspended and instead you can rubberband.

pavel-ch, 2019-09-03:
new Alt behavior: OK fine thank you.

kasemir, 2019-09-03:
I agree that having to press 'Alt' is a bit iffy, but it's kind of the natural extension of the general selection behavior: Click selects. Rubberband can select, but the first click to start a rubberband would select the group when you try to rubberband within a group. So we have alternate behavior to force rubberbanding by holding Alt, thus ignoring the select-click in the group.

Anyway, thanks for listing these issues. I think it resulted in fixing some unexplained widget location offsets that were there for a long time, but never enough spare time to fix them.

pavel-ch, 2019-09-05:
In the run to enhance Display Editor, I implemented the persistence of UI user settings in the editor, namely:
- Grid, Snap and Show Coordinates presets
- Divider positions between Tree, Edit and Property panes.
This is handy as for example we do not use "Snap" feature, so we do not have to switch it off always after activating editor.
The PR follows.

kasemir, 2019-09-05:
Yes, it's also good how you implemented that based on preferences, since these settings should carry over with the user, and not be persisted in the display file.

I'm somewhat unclear about the grid size. Snapping to the grid or not is clearly something that you might turn on/off while editing, and it's nice to remember your last choice when you edit the next file, or restart the tool. The grid size, however, is stored in the display file. So if you prefer to use "15x15" for a grid, you need to configure that each time you create a new display. And if you open an existing display, you get the grid size saved in that display. On the other hand, some displays may be designed for a 5x5 grid, others for a 20x20 grid, and it makes sense to then store that in the display file....

pavel-ch, 2019-09-05:
I am definitely accepting the grid size saved to the display file. This is a design choice related to the file itself. From my perspective, no need to persist it in the editor, which can in turn lead to a confusion. So no change, the current implementation is just fine.

pavel-ch, 2019-09-05:
In this regard:
> you need to configure that each time you create a new display

I would propose to make the template display path for
`org.csstudio.display.builder.editor.app.NewDisplayMenuEntry`
class changeable via settings.ini preferences. Currently it is set fixed to `examples:/initial.bob`.
This way we can choose a site or project specific template, including grid size but also titles, display size etc.

kasemir, 2019-09-05:
Excellent idea. Can you make that a preference?

pavel-ch, 2019-09-05:
yes  I try to do asap.

pavel-ch, 2019-10-23:
I implemented the Ctrl-D (duplicate) shortcut to the DisplayBuilderEditor. It still has some issues with Groups but these can be tuned later. Can I PR?

pavel-ch, 2020-02-05:
My users complain that when pasting widgets (or Group), widgets get off grid even when grid is on. Grid constraint has to be applied when pasting to editor with grid on.

pavel-ch, 2020-02-05:
Pressing the left mouse button (start selection gesture) on a display background with zoom level > 100 makes the display to shift (move the view) slightly left-up. The move seems being proportional to the zoom factor.
To reproduce: open a display in Editor, set zoom to 1000%, press left mouse, observe upper-left corner.
Looks like that there is the shift when a selection is active (visible). There is a gap initially (incorrect?) there, which disappears with an active selection.
Any idea why this is happening in this situation @kasemir ? Thanks.

pavel-ch, 2020-02-05:
There is a demand from users to add a function which would align selected widgets to grid (individually for each widget in a selection, apply the grid constraint). Useful when there is a misplacement to the grid.
I would add it at the bottom of "Align" menu, ok?

kasemir, 2020-02-05:
"Align to grid" sounds good.
Would it only round (x, y) to the closest grid point, or then also adjust width & height to have the bottom-right corner fall onto a grid point?

pavel-ch, 2020-02-05:
The intent is to round x, y. There may be reasons to keep the size off-grid (that is relatively usual if you need a center point, then you keep the size in odd numbers)

pavel-ch, 2020-02-21:
There was a demand from users here for the following context menu items:
- Expand All Tree Items in WidgetTree
- Collapse All Tree Items in WidgetTree
- Duplicate Widgets action in Display Editor (action itself already exists)
PR follows, hopefully you agree @kasemir

pavel-ch, 2020-03-06:
It would be helpful to see Widgets marked as Not Visible in the editor (grayed or an outline), or at least to see a (non) visibility mark in the Widget Tree.

pavel-ch, 2020-03-09:
Working on OPIs, I often end up with several displays in "main" area opened twice - in Editor and in Runtime (testing). Right now, I cannot quickly distinguish between Editor and Runtime instances: both show the display title in the Tab. I suggest to prepend [E] or [EDIT] in the Tab for Editor instances. Agree @kasemir ?

kasemir, 2020-03-09:
OK. I'd prefer '[Edit]' over just '[E]'.

pavel-ch, 2020-04-21:
It should be possible to Copy / Paste Properties from / to DisplayModel (display "background") in Editor

pavel-ch, 2020-07-28:
My users are asking for Widget Distribution functions (Horizontal, Vertical) with a predefined gap. I think these can be generally useful. Gaps are set via a dialog and stored in preferences for a next usage. PR follows.

pavel-ch, 2020-07-28:
Question:
My user complains that on a selected widget, the WH size label is visible while XY location label is not visible.
Is there any reason for inconsistency in showing/hiding thresholds between them?
In org.phoebus.ui.javafx.Tracker.java:477 and below
```
        locationLabel.setVisible(showLocationAndSize && ( width > 50 && height > 20 ));
        sizeLabel.setVisible(showLocationAndSize && ( width > 25 && height > 10 ));
```
In this case, we have height 15 widgets.
But as stated above, just a question.
Thanks, @kasemir

kasemir, 2020-07-28:
I think this is how Claudio originally implemented it. I usually turn those off anyway, so never noticed, but I can see that if the widget is small, only one of them will fit. At that point, I would agree that the size is somewhat more important than the position, but in any case you'd have to pick one.

[image]

pavel-ch, 2020-07-28:
I see. The issue is that the XY is hidden when not necessary:
[image]
Can I improve the conditions?

kasemir, 2020-07-28:
Sure, go ahead.

kasemir, 2020-07-28:
There are also cases where both are shown but shouldn't.

[image]

pavel-ch, 2020-07-30:
In the Display Builder Editor, there are several situation where a fast direct opening a plain text editor would be helpful:
- edit script file in Script dialog
- open BOB file as XML text for a fast find/replace
- etc.
Finding the file again and again in an external program is really annoying, users are complaining.

Phoebus already includes an external tools configuration support.
Can I add some buttons to the relevant dialogs or toolbars to allow open a suitable tool if configured?

georgweiss, 2020-07-30:
Agree, the ability to quickly edit the XML is very useful. We have users asking for ability to edit the ini file.

kasemir, 2020-07-30:
How do you open display files that you want to edit?
If you do this from within the Phoebus file browser, you can add a text editor right there using the "External applications" mechanism, 
https://github.com/ControlSystemStudio/phoebus/blob/master/core/framework/src/main/resources/workbench_preferences.properties#L23

[image]

kasemir, 2020-07-30:
> .. asking for ability to edit the ini file.

The  "External applications" example specifically includes 'ini' files:
```
# external_app_text=Text Editor,txt|dat|py|ini|db|xml|xsl|css|cmd|sh|st|log|out|md|shp,gedit
```

georgweiss, 2020-07-30:
Let me clarify: they want edit inside Phoebus. I basically fail to see the benefit of that, so configuring an external editor might be good enough.

kasemir, 2020-07-30:
Those who desire a tool that integrates/duplicates 150% of what you need can still try to build the Eclipse-based CS-Studio.
For Phoebus, we stated from the start that we would NOT include support for CVS, Subversion, Git, PyDev, XML Editor, ...
https://docs.google.com/document/d/11W52PRlsRjpIvP81HxUxxR9g180DHDByCohYQ9TQv7U

georgweiss, 2020-07-30:
Agree completely. I've already fenced off git integration, I will do the same with text editor for the ini file. External tool config should suffice.

pavel-ch, 2020-07-30:
Yes my suggestion is only about using existing external_app feature to open faster e.g. script from Script dialog or .bob xml from Display Editor toolbar or menu, just like opening the runtime. Not more. If not desired, I can still live without that.

kasemir, 2020-07-30:
OK on listing an external editor for any of 'xml', 'opi', 'bob' from the display editor. Menu may be better than toolbar since toolbar is already pretty full.

pavel-ch, 2020-09-11:
Group behavior in Editor, again:
Invisible Group (Tab) should NOT accept dropping of widgets on its area. This makes editing visually unpredictable. Widgets should be placed into Group Parent instead of the invisible Group.

pavel-ch, 2021-05-25:
Display Editor: My users are requesting a CrossHair mouse cursor in Editor (ala CAD), like the crosshair cursor in Databrowser3.
Good for measurements and alignment works in displays.
This is hopefully useful for others, too.
I prototyped that. Activation by pressing c key or by icon just like in databrowser. Always off at startup.
[image]
I tried to resolve issues about zoom/pan etc.
Code in my fork https://github.com/pavel-ch/phoebus  master.
I issue a PR.
Can you approve @kasemir ? Thanks.

pavel-ch, 2021-07-28:
I have got a request from users for a better Macro hierarchy overview in Display Editor. In the first phase, the request is targeting to a better visual recognition of widgets with macros defined inside. The solution proposed is to add an additional icon to such widgets in the Editor Widget Tree as follows:
[image]

Open issues:
- refresh after adding/deleting macros (now after a selection change)
- icon (now a random dollar sign icon, can be more pronounced)
- display model macros: for now, not indicated because of no item for Display in the tree

I prototyped that in my fork. Is that acceptable / can I PR @kasemir ?
Thank you.

closed 2023-03-14 by pavel-ch