ControlSystemStudio/phoebus #695: How to open a bob as dialog window?

forestmojo, 2019-06-19:
hi,

I want to implement a valve controls. since there is no widget of valve, so I use Polygon and Ellipse widget to draw a figure like a valve. I want to click it, then it pop a new bob file as dialog, then operator can adjust PID parameters.
Any idea or suggestions? thanks a lot for help.

kasemir, 2019-06-19:
If you want to create displays, looking through the examples is a good starting point.
05_actions.bob shows how you can place a transparent button on top of graphics, so clicking that region will then invoke an action to for example open a new display.
If you did look at that example and missed that fact that the region is clickable, that's exactly the point.
From the example:
"If you end up with many displays that have invisible buttons,
and operators need to guess where they can click,
you may want to look for professional help,
or a job in the video game industry."

forestmojo, 2019-06-19:
:-p. I use same idea as yours at first, but choose the wrong propertiy of button, now I change visible properity to transparent properity. that's the reason.

closed 2019-06-19 by forestmojo

pavel-ch, 2019-06-19:
I see the main point here:
> then it pop a new bob file as dialog

This is not possible at the moment. I am missing the feature, too, being a usual pattern to have a summary info about an object (valve...) in the picture and a detailed view in a popup dialog.
This needs a dedicated OpenDisplay action (OpenPopUp ...), which opens a special popup dialog window with bob file runtime, close to the originating widget position, just like a Dialog now.
Necessary is some support in Actions, Display Editor (actions dialog) and runtime.
And yes there are possible side effects which need to be handled:
- modality of the new window is a question, on the first try I would make the popup always modal
- what if the bob file in popup opens a new bob, in popup or with a regular Open
- etc.
I can make a prototype in a mid term, are you interested Kay (@kasemir )?

kasemir, 2019-06-19:
Popups are cute when working on a single display, thinking this display is all operators will ever use.
But operators may prefer to open several displays, and starting modal displays will then block access to all but that one popup.
Still, I could see adding popups as job security, because it later allows selling the desperately needed popup-blocker at a premium.

pavel-ch, 2019-06-19:
👍 yes this is one of cons. Searching for a forgotten modal popup on several screens is really annoying. But we are creating a tool, which should be "complete". How the tool is used is another story. But I can imagine (=I have seen) perfectly working scenarios using popup windows. If you like to have a user base "behind Epics", you have to imagine even small single-panel applications, for which this could be an appropriate solution.
Or, imagine an Opi Probe. Why not using a popup? Now, Probe covers the whole screen by default which is not what I expect, for example wanting to manipulate with a PV and observing a display object reaction. Etc.

katysaintin, 2022-02-09:
Hello, I'm searching a way to open a Bob as Popup modal dialog Phoebus. Is there a option that allows this feature ?

kasemir, 2022-02-09:
There's not.  You can open a display as a new window, but not as a modal dialog.
There is the option to use the ScriptUtil to open basic dialogs, see examples, scripts, script util, dialogs, but I already regret offering that because it has led to users overlooking the dialog and then complaining that the whole application is "frozen".

katysaintin, 2022-02-16:
Dear @kasemir 

We succeed to set the stage alwaysOnTop and not modal.

But my user want also some popup in modal mode so 
I call the following function in my script
stage.initModality(Modality.APPLICATION_MODAL)

=> I have an exception 
Exception in thread "JavaFX Application Thread" java.lang.IllegalStateException: Cannot set modality once stage has been set visible at javafx.stage.Stage.initModality(Stage.java:516)

Even If I do something like this 
stage.hide()
stage.initModality(Modality.APPLICATION_MODAL)
stage.show()

Do you have and Idea to do something to disable the other views when a Dialog is open ?
Thank you for your help.

kasemir, 2022-02-16:
Well, BOY had the option to open new displays in a "standalone window", a plain SWT window that's not integrated into Eclipse. That of course has no equivalence in Phoebus nor the web runtime. A modal dialog has no generally supported equivalent for the web runtime. Plus, imagine you're in such a modal dialog, and then right-click on a PV to open in the data browser or probe or ..., what's that supposed to do? So bottom line, sorry, I don't see a good way to support this in a consistent, future-proof manner.

legrandarnaud, 2022-02-17:
Hi Kay,

I am Katy's colleague.

on our application, i want a popup window with the possibility to write pvs in the popup, when the popup is on, we needn't to write the pvs on the main display.

on the picture, the script put good the popup above the main display but I have always the possibility to write the pvs on main display. 

thank you for the return

Arnaud

[image]

kasemir, 2022-02-17:
Again, popups are not supported, and I don't see how they can be supported in a sensible way: Context menus won't make sense, can't implement that in the web runtime, would be restricting portability to future display tools. If you prefer a display tool where you can programmatically do anything you want, without being restricted by a display file format that's meant to be portable across several tools, see https://slaclab.github.io/pydm-tutorial/