ControlSystemStudio/phoebus #705: Phoebus fails on start with X11 issue

kasemir, 2019-06-24:
Sometimes, starting Phoebus on Linux aborts with this error:
```
[xcb] Unknown request in queue while dequeuing
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
```

Reason is most likely some call to JavaFX API off the UI thread, but unclear where.

kasemir, 2020-01-08:
Still occasionally happening...

willrogers, 2020-10-19:
This is happening almost every time I open a databrowser plot in CS-Studio when building against the latest Eclipse version. This suggests that it's not just JavaFX, but also that it's not just SWT. Hm.

willrogers, 2020-10-19:
It sounds like this one is not particularly associated with the databrowser, is that correct?

kasemir, 2020-10-19:
I think it's unrelated to data browser, 100% related to JavaFX.
For Eclipse-based CSS, data browser may just happen to be the first JavaFX panel that you open.

kasemir, 2020-10-19:
Any difference for GTK 2 vs 3? `-Djdk.gtk.version=2` vs. 3?

willrogers, 2020-10-19:
I'm using the original databrowser, SWT based. Is there any JavaFX involved?

willrogers, 2020-10-19:
This is all on GTK3. I've never seen anything on the older versions of CS-Studio and I can't run GTK2 on this newest one.

willrogers, 2020-10-19:
I expect there is a bug in SWT, but unless we pin it down a bit no-one's going to fix it. And if we can pin it down then maybe we can avoid it.