ControlSystemStudio/phoebus #715: Support for referencing elements in a pvData arrays

shroffk, 2019-06-27:
While the current pva support allows users to request individual fields from a pvData structure there are use cases where it would be useful to reference an element in a array.

While it is not possible to request element level updates from pva, the pva client could add this functionality to 
1. make it easier for the user by not requiring them to process the pvStructure.
2. make things a bit more efficient by throwing away the rest of the unwanted data before converting it to vtypes and passing it on to the user application.

kasemir, 2019-06-27:
The Data Browser has somewhat quirky element support.
Each 'Trace' has an 'Index' to access a waveform element.
This was added by KEK, I believe to look at basically structured data shoehorned into CA waveforms, since CA has no other structure support, to then plot a structure element (= array element).

Could add the same "some_pv[index]" support to "ca://" PVs, so "ca://some_waveform[i]" would also fetch an array element, removing need for "index" in Data Browser traces.

shroffk, 2019-06-27:
Yes, I think it would be great to add it to ca too...it might be easier than pva

currently, I have support for 
```
pva://pv_array[i]  ==> pva://pv_array.value[i] 
pva://pv_struct/array_field[i]
pva://pv_struct?request(field=some.array_field[i])
```

There is some duplication in the code which can be optimized but I thought it best to handle each of the above situations separately.

TODO:

it would be nice to have support for  
`pva://...[i].some_sub_field`   
&&  
`pva://...[-1]  ==> last element`

kasemir, 2019-06-27:
For the sake of completeness I can argue for some form of sub-array support like `[start:stop]` or even `[start:stop:step]`.
Of course using the python slice notation where most end users forget that 'stop' is exclusive, so they're one array element short of expectations and you can conveniently shame them for not being clever enough to understand python.

jjaraalm, 2021-05-23:
Any updates/plans for supporting PVA structure arrays? `pva://...[i].some_sub_field` would be very nice to have.

Structure of arrays is an alternative that I might fall back on, but it's not as nice.

jjaraalm, 2021-05-23:
Ah, I see. There is also an issue with field names for structure arrays.

In `org.phoebus.pv.pva.PVAStructureHelper.getVType`, if the structure array is not in a `value` field, the first block gets triggered, but `org.phoebus.pv.pva.Decoders.decodeArray` cannot handle structure arrays.

Test PVs:

```
test structure 
    structure[] value
        structure 
            int x 0
        structure 
            int x 0

test2 structure 
    structure[] v
        structure 
            int x 0
        structure 
            int x 0
```

I can do `pva://test[0]` or `pva://test/value[0]` in Phoebus, but `pva://test2/v[0]` fails.