ControlSystemStudio/phoebus #720: Phoebus versioning

shroffk, 2019-06-28:


kasemir, 2019-06-28:
In favor of 4.6 or 5, anything that suggests is a continuation of CS-Studio.
Maybe 7 since it's a big step; goes with EPICS 7.

Discussing version numbers is a terrific Friday topic!

berryma4, 2019-06-28:
I vote 5 or 7 :)

georgweiss, 2019-06-28:
What about org.epics:pbrawclient:0.0.8? Can't find it on Maven central, guess that's why it is handled separately in module install-jars. Can we get rid of that by building it and push it to Maven Central?

kasemir, 2019-06-28:
Please add the Oracle JDBC jar to Maven central as well.

shroffk, 2019-06-28:
hmm...I am approach murali about adding pbrawclinet

the oracle jdbc is a different story

shroffk, 2019-06-28:
my vote is for 4.6 since we say that it is all part of the CS-Studio 4.6 release

closed 2022-08-10 by shroffk