ControlSystemStudio/phoebus #726: Cannot save *.bcf file

MaticPogacnik, 2019-07-04:
Hi,

I am trying to create bcf file with predefined classes.
But when I try to save it i get an error that a file does not have suggested file extension. And I am unable to save it.

[image]

It doesn't matter if I click Cancel or OK, either way I get dialog to select a name again and file is not saved.

Thanks,
Matic

kasemir, 2019-07-04:
I can confirm that for now you'll need to save it as a *.bob file and then rename.
Will fix this next week.

kasemir, 2019-07-04:
You can also try to fix this and submit a pull request.
At first glance, the required changes should be limited to the code around this section in https://github.com/shroffk/phoebus/blob/master/app/display/editor/src/main/java/org/csstudio/display/builder/editor/app/DisplayEditorInstance.java#L315:
```
 final File proper = ModelResourceUtil.enforceFileExtension(file, DisplayModel.FILE_EXTENSION);
```

Right now it always uses the *.bob `DisplayModel.FILE_EXTENSION`.
Should check `model.isClassModel()` to determine if maybe the bcf extension from 
`WidgetClassSupport.FILE_EXTENSION` is required.

closed 2019-07-11 by MaticPogacnik

kasemir, 2019-07-11:
Thanks for finding this and then fixing it.
Please continue ;-)