ControlSystemStudio/phoebus #730: Authentication support for ElasticSearch for Alarm Logger and logging-UI

tanviash, 2019-07-08:
@shroffk Hi Kunal, we need to support basic authentication for elasticsearch for alarm logger service and logging UI for added security. Alarm logger and logging-ui currently does not support any auth for elasticsearch.

shroffk, 2019-07-08:
Hi @tanviash 

I am not sure I understand the reason here...

1. The logger service can be a standalone service, running independently on a separate machine in an isolated environment. All the service does is record alarm messages into the elastic backend.

2. The logger ui only allows for reading, it does not allow users to change or add data.

In both these cases how would having authentication and authorization add to security? is searching for alarm messages a security concern at FRIB?

tanviash, 2019-07-08:
@shroffk ElasticSearch cluster over here, like any other database, is required to have basic authentication. This cluster is shared by many applications (non-alarms related) and alarm logger is one of them. We really do not have a separate elasticsearch setup for alarms.

tanviash, 2019-07-08:
@shroffk I can work on adding this support and have you code-review it if that's okay.

shroffk, 2019-07-08:
sure sure...

There is the elastic security pack
https://www.elastic.co/guide/en/elastic-stack-overview/6.8/setting-up-authentication.html

we could look at this for address some of the AA needs. We should ensure that the easy of deployment and the stability and functionality of the services are not affected.

shroffk, 2020-06-04:
@tanviash 
I think we had discussed about adding a rest service on top of alarm-logging. 
That could also add the authentication needed.

Should we close this and open a new Issue about adding a rest interface on top of alarm-logging. This should also help with future upgrades and migrations of elastic search.

tanviash, 2020-06-04:
@shroffk Sure, I'm fine with that.

closed 2020-06-04 by tanviash