ControlSystemStudio/phoebus #743: Annotations "drifting"

mdavidsaver, 2019-07-18:
More accurately, annotations not moving when plot data updates.  In the following plot the line moves up on successive updates.  I noticed that an annotation doesn't update until I click and drag.  Then it jumps back to the line.

[image]

kasemir, 2019-07-24:
True.
The annotations were first added to the data browser, looking at archived data.
Historic samples taken at some point in time tend to not change their value, which is why the annotations' values are only determined when an annotation is positioned, and not recomputed for each value update.

mdavidsaver, 2019-07-24:
To be clear, this is a very minor issue for me.

kasemir, 2019-07-24:
Understood. Same issue with the cursor markers, they only get values when moved.
Re-determining values of annotations and cursor markers on each value update (actually: graph redraw, could skip a few value updates) isn't free, so need to look at it some more when I find the time.

kasemir, 2019-07-29:
Have an update that moves the annotation to the 'current' value.
It's great fun trying to 'grab' an annotation to adjust its position when it keeps moving up/down with the data.

kasemir, 2019-08-02:
Annotations now move up/down with the data.

closed 2019-08-02 by kasemir