ControlSystemStudio/phoebus #747: Alarm Logger Exception in logs

tanviash, 2019-07-23:
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]: 2019-07-22 13:03:19.820 ERROR 26997 --- [-StreamThread-2] o.a.k.s.e.LogAndFailExceptionHandler     : Exception caught during Deserialization, taskId: 0_0, topic: alarm_diag, partition: 0, offset: 35778
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]: [1B blob data]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]: org.apache.kafka.common.errors.SerializationException: com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field "latch" (class org.phoebus.applications.alarm.messages.AlarmMessage), not marked as ignorable (21 known properties: "enabled", "delay", "message", "host", "user", "time", "annunciating", "displays", "filter", "description", "actions", "latching", "key", "current_message", "current_severity", "commands", "count", "value", "guidance", "severity", "mode"])
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:  at [Source: (byte[])"{"severity":"MAJOR","latch":true,"message":"STATE_ALARM","value":"NOK","time":{"seconds":1563814941,"nano":719233520},"current_severity":"MAJOR","current_message":"STATE_ALARM"}"; line: 1, column: 33] (through reference chain: org.phoebus.applications.alarm.messages.AlarmMessage["latch"])
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]: Caused by: com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field "latch" (class org.phoebus.applications.alarm.messages.AlarmMessage), not marked as ignorable (21 known properties: "enabled", "delay", "message", "host", "user", "time", "annunciating", "displays", "filter", "description", "actions", "latching", "key", "current_message", "current_severity", "commands", "count", "value", "guidance", "severity", "mode"])
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:  at [Source: (byte[])"{"severity":"MAJOR","latch":true,"message":"STATE_ALARM","value":"NOK","time":{"seconds":1563814941,"nano":719233520},"current_severity":"MAJOR","current_message":"STATE_ALARM"}"; line: 1, column: 33] (through reference chain: org.phoebus.applications.alarm.messages.AlarmMessage["latch"])
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException.from(UnrecognizedPropertyException.java:60) ~[jackson-databind-2.9.6.jar!/:2.9.6]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at com.fasterxml.jackson.databind.DeserializationContext.handleUnknownProperty(DeserializationContext.java:822) ~[jackson-databind-2.9.6.jar!/:2.9.6]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at com.fasterxml.jackson.databind.deser.std.StdDeserializer.handleUnknownProperty(StdDeserializer.java:1152) ~[jackson-databind-2.9.6.jar!/:2.9.6]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownProperty(BeanDeserializerBase.java:1589) ~[jackson-databind-2.9.6.jar!/:2.9.6]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownVanilla(BeanDeserializerBase.java:1567) ~[jackson-databind-2.9.6.jar!/:2.9.6]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:294) ~[jackson-databind-2.9.6.jar!/:2.9.6]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:151) ~[jackson-databind-2.9.6.jar!/:2.9.6]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4013) ~[jackson-databind-2.9.6.jar!/:2.9.6]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3091) ~[jackson-databind-2.9.6.jar!/:2.9.6]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.phoebus.applications.alarm.messages.MessageParser.deserialize(MessageParser.java:54) ~[app-alarm-model-4.6.0-SNAPSHOT.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.common.serialization.ExtendedDeserializer$Wrapper.deserialize(ExtendedDeserializer.java:65) ~[kafka-clients-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.common.serialization.ExtendedDeserializer$Wrapper.deserialize(ExtendedDeserializer.java:55) ~[kafka-clients-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.streams.processor.internals.SourceNode.deserializeValue(SourceNode.java:63) ~[kafka-streams-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.streams.processor.internals.RecordDeserializer.deserialize(RecordDeserializer.java:66) ~[kafka-streams-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.streams.processor.internals.RecordQueue.addRawRecords(RecordQueue.java:97) ~[kafka-streams-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.streams.processor.internals.PartitionGroup.addRawRecords(PartitionGroup.java:117) ~[kafka-streams-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.streams.processor.internals.StreamTask.addRecords(StreamTask.java:638) ~[kafka-streams-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.streams.processor.internals.StreamThread.addRecordsToTasks(StreamThread.java:936) ~[kafka-streams-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.streams.processor.internals.StreamThread.runOnce(StreamThread.java:831) ~[kafka-streams-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.streams.processor.internals.StreamThread.runLoop(StreamThread.java:767) ~[kafka-streams-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]:         at org.apache.kafka.streams.processor.internals.StreamThread.run(StreamThread.java:736) ~[kafka-streams-2.0.0.jar!/:na]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]: [1B blob data]
Jul 22 13:03:19 alarm-diag phoebus-alarm-logger[26996]: 2019-07-22 13:03:19.864  WARN 26997 --- [-StreamThread-2] org.apache.kafka.streams.KafkaStreams    : stream-client [streams-alarm_diag-alarm-messages-7c0ca6c3-eebf-421b-a62a-39b0377a3c54] All stream threads have died. The instance will be in error state and should be closed.

kasemir, 2019-07-23:
See #553 :
Before Feb.5, the alarm server only sent updates, and the alarm logger had to figure out if this is a latching channel that right now enters the latched state.
With that pull request, the alarm server added a "latch" tag to the message when this state update indicates an alarm entering the latched state.

Looks like org.phoebus.applications.alarm.messages.AlarmMessage needs to be updated to include a 'boolean latch'

tanviash, 2019-07-23:
@kasemir Shouldn't we be just renaming the existing 'latching' field org.phoebus.applications.alarm.messages.AlarmMessage to 'latch'?

kasemir, 2019-07-23:
latching: Is this PV meant to latch, or will it always update to the current severity?

latch: Flag that indicates that a PV configured to be latching actually latches right now.

kasemir, 2019-07-23:
In other words, `latching` is a configuration message element, `latch` is a state update element.

tanviash, 2019-07-23:
@shroffk I just pulled the latest code and built alarm logger. But when I start alarm logger it starts to throw errors.

2019-07-23 17:37:47.345 ERROR 24594 --- [-StreamThread-1] o.a.k.s.p.i.AssignedStreamsTasks         : stream-thread [streams-alarm_local-alarm-messages-56ebbdd3-111e-4b2b-864e-28e59becbfc0-StreamThread-1] Failed to process stream task 0_0 due to the following error:

org.elasticsearch.ElasticsearchStatusException: Elasticsearch exception [type=illegal_argument_exception, reason=Rejecting mapping update to [alarm_local_alarms_config_2019-07-01] as the final mapping would have more than 1 type: [alarm_config, alarm]]
	at org.elasticsearch.rest.BytesRestResponse.errorFromXContent(BytesRestResponse.java:177) ~[elasticsearch-6.4.2.jar!/:6.4.2]
	at org.elasticsearch.client.RestHighLevelClient.parseEntity(RestHighLevelClient.java:1406) ~[elasticsearch-rest-high-level-client-6.4.2.jar!/:6.4.2]
	at org.elasticsearch.client.RestHighLevelClient.parseResponseException(RestHighLevelClient.java:1382) ~[elasticsearch-rest-high-level-client-6.4.2.jar!/:6.4.2]
	at org.elasticsearch.client.RestHighLevelClient.performRequest(RestHighLevelClient.java:1269) ~[elasticsearch-rest-high-level-client-6.4.2.jar!/:6.4.2]
	at org.elasticsearch.client.RestHighLevelClient.performRequest(RestHighLevelClient.java:1240) ~[elasticsearch-rest-high-level-client-6.4.2.jar!/:6.4.2]
	at org.elasticsearch.client.RestHighLevelClient.performRequestAndParseEntity(RestHighLevelClient.java:1223) ~[elasticsearch-rest-high-level-client-6.4.2.jar!/:6.4.2]
	at org.elasticsearch.client.RestHighLevelClient.index(RestHighLevelClient.java:598) ~[elasticsearch-rest-high-level-client-6.4.2.jar!/:6.4.2]
	at org.phoebus.alarm.logging.ElasticClientHelper.indexAlarmStateDocument(ElasticClientHelper.java:92) ~[classes!/:na]
	at org.phoebus.alarm.logging.AlarmMessageLogger.lambda$processAlarmStateStream$6(AlarmMessageLogger.java:170) ~[classes!/:na]
	at org.apache.kafka.streams.kstream.internals.KStreamPeek$KStreamPeekProcessor.process(KStreamPeek.java:42) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:143) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:126) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:90) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.kstream.internals.KStreamFilter$KStreamFilterProcessor.process(KStreamFilter.java:43) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:143) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:126) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:90) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.kstream.internals.KStreamTransform$KStreamTransformProcessor.process(KStreamTransform.java:59) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:143) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:126) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorContextImpl.forward(ProcessorContextImpl.java:90) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.kstream.internals.KStreamPassThrough$KStreamPassThroughProcessor.process(KStreamPassThrough.java:33) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode$1.run(ProcessorNode.java:50) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode.runAndMeasureLatency(ProcessorNode.java:244) ~[kafka-streams-2.0.0.jar!/:na]
	at org.apache.kafka.streams.processor.internals.ProcessorNode.process(ProcessorNode.java:133) ~[kafka-streams-2.0.0.jar!/:na]
[... cut ...]
	Caused by: org.elasticsearch.client.ResponseException: method [POST], host [http://ctlsrch01.nscl.msu.edu:9200], URI [/alarm_local_alarms_config_2019-07-01/alarm?timeout=1m], status line [HTTP/1.1 400 Bad Request]
{"error":{"root_cause":[{"type":"illegal_argument_exception","reason":"Rejecting mapping update to [alarm_local_alarms_config_2019-07-01] as the final mapping would have more than 1 type: [alarm_config, alarm]"}],"type":"illegal_argument_exception","reason":"Rejecting mapping update to [alarm_local_alarms_config_2019-07-01] as the final mapping would have more than 1 type: [alarm_config, alarm]"},"status":400}

shroffk, 2019-07-26:
> {"error":{"root_cause":[{"type":"illegal_argument_exception","reason":"Rejecting mapping update to [alarm_local_alarms_config_2019-07-01] as the final mapping would have more than 1 type: [...]

this is different form the latch problem.

I tested the create template and index files with elastic 6.3.1 and 6.8.1...they worked fine on both.

Now, about the new field latch...I will try to recreate the problem above

shroffk, 2019-07-26:
@tanviash based on our email exchange. can we safely conclude that the elastic index creation problem is now solved.

tanviash, 2019-07-26:
@shroffk Hi Kunal, Yes the problem with multiple mapping went away when I deleted kafka topics and created them again. However, I'm not sure if it will not come up again. The issue with unrecognized field "latch" still is there.

shroffk, 2019-07-26:
ok...I see why this is happening...I should have a patch soon

shroffk, 2019-07-26:
Can you check the Issue_747 branch
Both the above errors should be solved.

shroffk, 2019-07-29:
Verified the services and app's continue to work after the refactoring of the alarm service's kafka messages
[image]

shroffk, 2019-08-05:
@tanviash can we close this?