ControlSystemStudio/phoebus #748: Cannot resolve imports in Appliance Archive

DorianHas, 2019-07-24:
Hi there, !

My config :
- CentOs 7
- JDK 12
- Latest Phoebus version 
- Intellij 2019.1.3
- Eclipse 2019-06 (4.12.0)

Project import, from Maven in Eclipse and Intellij, went well so I tried to compile it. Some dependencies related to Appliance Archieve Reader cannot be resolved.

Fun fact, with maven comand lines, it works well. It might be something I missed during Maven import.

[image]

Here's Intellij log :
[Issue Appliance Archive.txt](https://github.com/shroffk/phoebus/files/3426521/Issue.Appliance.Archive.txt)

Information:java: Errors occurred while compiling module 'app-databrowser'
Information:javac 12.0.1 was used to compile java sources
Information:Modules "app-display-convert-medm", "product" were fully rebuilt due to project configuration/dependencies changes
Information:24/07/2019 12:43 - Build completed with 97 errors and 36 warnings in 4 m 17 s 29 ms
T:\Phoebus\src\core\util\src\test\java\org\phoebus\util\time\TimeParserTest.java
    Warning:Warning:line (28)java: getInstant(java.lang.String) in org.phoebus.util.time.TimeParser has been deprecated
    Warning:Warning:line (39)java: parseDuration(java.lang.String) in org.phoebus.util.time.TimeParser has been deprecated
    Warning:Warning:line (42)java: parseDuration(java.lang.String) in org.phoebus.util.time.TimeParser has been deprecated
    Warning:Warning:line (46)java: parseDuration(java.lang.String) in org.phoebus.util.time.TimeParser has been deprecated
    Warning:Warning:line (49)java: parseDuration(java.lang.String) in org.phoebus.util.time.TimeParser has been deprecated
    Warning:Warning:line (52)java: parseDuration(java.lang.String) in org.phoebus.util.time.TimeParser has been deprecated
    Warning:Warning:line (72)java: parseDuration(java.lang.String) in org.phoebus.util.time.TimeParser has been deprecated
    Warning:Warning:line (75)java: parseDuration(java.lang.String) in org.phoebus.util.time.TimeParser has been deprecated
    Warning:Warning:line (116)java: parseDuration(java.lang.String) in org.phoebus.util.time.TimeParser has been deprecated
T:\Phoebus\src\core\framework\src\main\java\org\phoebus\framework\preferences\FileSystemPreferences.java
    Warning:Warning:line (720)java: Long(long) in java.lang.Long has been deprecated
T:\Phoebus\src\core\framework\src\main\java\org\phoebus\framework\preferences\FilePreferencesXmlSupport.java
    Warning:Warning:line (270)java: Integer(int) in java.lang.Integer has been deprecated
T:\Phoebus\src\core\framework\src\main\java\org\phoebus\framework\selection\SelectionUtil.java
    Information:Information:java: Some input files use unchecked or unsafe operations.
    Information:Information:java: Recompile with -Xlint:unchecked for details.
T:\Phoebus\src\app\logbook\ui\src\main\java\org\phoebus\logbook\ui\LogEntryTableApp.java
    Warning:Warning:line (46)java: canOpenResource(java.lang.String) in org.phoebus.framework.spi.AppResourceDescriptor has been deprecated
T:\Phoebus\src\app\logbook\ui\src\main\java\org\phoebus\logbook\ui\LogEntryCalenderApp.java
    Warning:Warning:line (46)java: canOpenResource(java.lang.String) in org.phoebus.framework.spi.AppResourceDescriptor has been deprecated
T:\Phoebus\src\app\logbook\ui\src\main\java\org\phoebus\logbook\ui\LogEntryTableViewController.java
    Information:Information:java: Some input files use unchecked or unsafe operations.
    Information:Information:java: Recompile with -Xlint:unchecked for details.
T:\Phoebus\src\app\logbook\inmemory\src\main\java\org\phoebus\applications\logbook\InMemoryLogClient.java
    Warning:Warning:line (225)java: findLogById(java.lang.Long) in org.phoebus.logbook.LogClient has been deprecated
    Warning:Warning:line (135)java: set(java.util.Collection<org.phoebus.logbook.LogEntry>) in org.phoebus.logbook.LogClient has been deprecated
T:\Phoebus\src\app\log-configuration\src\main\java\org\phoebus\applications\utility\LoggingConfiguration.java
    Information:Information:java: /T:/Phoebus/src/app/log-configuration/src/main/java/org/phoebus/applications/utility/LoggingConfiguration.java uses unchecked or unsafe operations.
    Information:Information:java: Recompile with -Xlint:unchecked for details.
T:\Phoebus\src\app\display\convert-medm\src\main\java\org\csstudio\utility\adlparser\fileParser\widgetParts\ADLConnected.java
    Warning:Warning:line (70)java: Integer(int) in java.lang.Integer has been deprecated
    Warning:Warning:line (71)java: Integer(int) in java.lang.Integer has been deprecated
T:\Phoebus\src\app\display\convert-medm\src\main\java\org\csstudio\utility\adlparser\fileParser\widgetParts\ADLObject.java
    Warning:Warning:line (219)java: Integer(int) in java.lang.Integer has been deprecated
    Warning:Warning:line (220)java: Integer(int) in java.lang.Integer has been deprecated
    Warning:Warning:line (221)java: Integer(int) in java.lang.Integer has been deprecated
    Warning:Warning:line (222)java: Integer(int) in java.lang.Integer has been deprecated
T:\Phoebus\src\app\display\convert-medm\src\main\java\org\csstudio\utility\adlparser\fileParser\widgetParts\ADLBasicAttribute.java
    Warning:Warning:line (153)java: Integer(int) in java.lang.Integer has been deprecated
    Warning:Warning:line (154)java: Integer(int) in java.lang.Integer has been deprecated
T:\Phoebus\src\app\display\convert-medm\src\main\java\org\csstudio\utility\adlparser\fileParser\widgetParts\ADLLimits.java
    Warning:Warning:line (38)java: Float(float) in java.lang.Float has been deprecated
    Warning:Warning:line (40)java: Float(float) in java.lang.Float has been deprecated
    Warning:Warning:line (42)java: Float(float) in java.lang.Float has been deprecated
T:\Phoebus\src\app\display\convert-medm\src\main\java\org\csstudio\utility\adlparser\fileParser\widgetParts\ADLPlotcom.java
    Warning:Warning:line (44)java: Integer(int) in java.lang.Integer has been deprecated
    Warning:Warning:line (45)java: Integer(int) in java.lang.Integer has been deprecated
[... 235 lines cut ...]
  location: class org.phoebus.archive.reader.appliance.ApplianceMeanValueIterator
    Error:Error:line (77)java: cannot find symbol
  symbol:   class GenMsgIterator
  location: class org.phoebus.archive.reader.appliance.ApplianceMeanValueIterator
    Error:Error:line (80)java: cannot find symbol
  symbol:   class PayloadInfo
  location: class org.phoebus.archive.reader.appliance.ApplianceMeanValueIterator
    Error:Error:line (81)java: cannot find symbol
  symbol:   class EpicsMessage
  location: class org.phoebus.archive.reader.appliance.ApplianceMeanValueIterator
    Error:Error:line (107)java: cannot find symbol
  symbol:   variable PayloadType
  location: class org.phoebus.archive.reader.appliance.ApplianceMeanValueIterator
    Error:Error:line (108)java: cannot find symbol
  symbol:   variable PayloadType
  location: class org.phoebus.archive.reader.appliance.ApplianceMeanValueIterator
    Error:Error:line (109)java: cannot find symbol
  symbol:   variable PayloadType
  location: class org.phoebus.archive.reader.appliance.ApplianceMeanValueIterator
    Error:Error:line (110)java: cannot find symbol
  symbol:   variable PayloadType
  location: class org.phoebus.archive.reader.appliance.ApplianceMeanValueIterator
    Error:Error:line (111)java: cannot find symbol
  symbol:   variable PayloadType
  location: class org.phoebus.archive.reader.appliance.ApplianceMeanValueIterator

kasemir, 2019-07-24:
You're missing `pbrawclient-0.0.8.jar` on the classpath.
The file is in dependencies/install-jars/lib/pbraw/pbrawclient-0.0.8.jar .

```
mvn clean verify  -f dependencies/pom.xml
```
collects all dependencies into `dependencies/phoebus-target/target/lib`, from which the `ant` and `eclipse` setups get their dependencies.

shroffk, 2019-07-24:
Hi, 
The pbraw library is not available in maven so we have to manually install it.
https://github.com/shroffk/phoebus/tree/master/dependencies/install-jars
The above project does just that...you can try to run the install on the it and see if intellij picks up the newly installed pbraw library

shroffk, 2019-07-24:
I should ask murali if he is willing to publish this library and save us this confusion.

DorianHas, 2019-07-25:
Perfect ! adding it to classpath solved the issue

closed 2019-07-25 by DorianHas