ControlSystemStudio/phoebus #749: How to configure OLog client

MaticPogacnik, 2019-07-26:
Hi,

I am trying to configure OLog client inside settings.ini file as :
`org.phoebus.logbook.ui/logbook_factory=olog
`
But I'm getting an warning:
`WARNING [org.phoebus.ui.application.PhoebusApplication] Cannot locate logbook factory 'olog'
`
Is there something I'm doing wrong?
And how can I configure URL and credentials?
My service is running on: http://10.72.10.92:8080/Olog/index.html

Thanks!!!
Lp, Matic

shroffk, 2019-07-26:
Currently the configuration is handled but the olog.properties files
You can place this file in the user home directory or for system wide settings you can put it under
/etc/olog.properties on linux or /olog.properties for windows

e.g. olog.properties files
```
olog_url=https://localhost:8181/Olog/resources
#Provide a path to the trustStore.jks 
#By default trustStore packaged with the jar will be used
#trustStore=/complete/path/to/java/trustStore
#trustPass=changeit
raw_html_logging=off
username=admin
password=XXXXXXX
```

TODO: 
@shroffk needs to integrate olog properties with phoebus preferences

kasemir, 2019-07-28:
>  WARNING Cannot locate logbook factory 'olog' 

Are you using the SNS version? It doesn't include the olog support, since we're using a different logbook.
Check if there's a file lib/*olog*.jar in your product to assert that it includes the olog support.

MaticPogacnik, 2019-07-28:
Thanks for fast responses Kay!

I cloned master branch from github and run: mvn clean install
Checked lib folder now and I cant find any *olog* files. Here is the output of search inside the phoebus folder:
`[user@host721057 phoebus]$ find . -name "*olog*"
./app/logbook/olog
./app/logbook/olog/src/main/java/org/phoebus/olog
./app/logbook/olog/target/classes/org/phoebus/olog
./app/logbook/olog/target/app-logbook-olog-4.6.0-SNAPSHOT.jar
`
How can can I include olog support to the build?

Thanks!

shroffk, 2019-07-28:
https://github.com/shroffk/nsls2-phoebus
Here is an example nsls2-product which does include the olog logbook library.

shroffk, 2019-07-28:
As far as I am aware, the common product also includes olog but it is not the default supported logbook.

kasemir, 2019-07-29:
> the common product also includes olog

I don't think it does.
The common product pom, https://github.com/shroffk/phoebus/blob/master/phoebus-product/pom.xml, lists logbook-inmemory, not olog.

I suggest adding it to the common product.

MaticPogacnik, 2019-07-30:
This is very useful information.
I had to include olog support in product pom and create olog.properties in home directory.
Then everything appeared to work (I had configured Logbooks from the service available), but when I tried to submit a log, I got an error:
`2019-07-30 14:28:50 WARNING [org.phoebus.framework.jobs] Job 'Submit Log Entry' failed
com.sun.jersey.api.client.UniformInterfaceException: POST https://10.72.10.92:8181/Olog/resources/logs returned a response status of 400 Bad Request
        at org.phoebus.olog.api.OlogClient.set(OlogClient.java:452)
        at org.phoebus.olog.api.OlogClient.set(OlogClient.java:427)
        at org.phoebus.logbook.ui.write.LogEntryModel.lambda$submitEntry$1(LogEntryModel.java:493)
        at org.phoebus.framework.jobs.Job.execute(Job.java:50)
        at org.phoebus.framework.jobs.JobManager.execute(JobManager.java:54)
        at org.phoebus.framework.jobs.JobManager.lambda$schedule$0(JobManager.java:45)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:834)
`
I captured the JSON content that gets send to the service and tried to do it manually with 3rd party REST client. This way I got more information. Looks like Phoebus was sending some unknown fields like title and so. When I removed all the fields on which olog service complained, log was successfully saved. 

War file I am using is: https://github.com/lnls-sirius/docker-olog-server/tree/master/bin. In my opinion this is due to version incompatibility as this war file is probably fairly old. 
Was my assumption correct? Is there minimum required version which I need?

Now I am trying to compile olog service (with little success till now).

kasemir, 2019-07-30:
I don't use the olog and can't help with its setup, but it's correct that the original olog didn't support 'title'. I'm a bit surprised that it complains about submissions with unknown fields instead of simply ignoring them. Still, I think your best path forward is to look for the newer olog that supports entries with title.

shroffk, 2019-07-30:
Hi @MaticPogacnik 

Ok, I was looking over the code in phoebus and I think you are right...the phoebus client was updated in preperation of the new olog server based on elastic search and mongo gridfs
https://github.com/shroffk/olog-es
Sorry...I don't have installation instruction for this service as yet

We have 2 options
1. I can refactor the olog client in phoebus to work with the old service :( 
2. We can work on trying to completing the new olog-es service.

shroffk, 2019-08-05:
@MaticPogacnik 

https://github.com/shroffk/phoebus/pull/757
In the latest PR I have added support for the old olog. You should be able to create and view log entries from the old olog service using the app-logbook-olog module.

The service url configuration is also now merged with phoebus preferences so you can use a settings.ini 

```
# --------------------------------------
# Package org.phoebus.olog.api
# --------------------------------------

# The olog url
olog_url=localhost:9092

# User credentials for olog
username=user
password=****

```
There are a few TODOs:
like support for properties
support for olog levels in the create logbook dialog
optimizing the fetching of attachments

The support for the new es based logbook is now it's own stand alone module (olog-es)

MaticPogacnik, 2019-08-05:
Hi @shroffk,

First sorry I didn't respond yet, was waiting for members of my team to come back (they are absent) so we can make decision together, which way we want to go. 

And thanks for the effort you put into this already. My first personal idea was also to make seperate support for each version.

Will test it as soon as possible.

Best regards

MaticPogacnik, 2019-08-13:
I made some tests and things that are important to us seem to work. We only have a problem with attachments (images) as they don’t get sent to the server.  
Easily including a screenshot of a display is a nice feature (What do they say?;-A picture is worth a thousand words. 😊 ) 
How much effort do you think it would take to implement this?
Thanks!

shroffk, 2019-08-19:
@MaticPogacnik can you test this patch

MaticPogacnik, 2019-08-20:
Thank you @shroffk . Its working now.

But there's a lot of output (also looks like binary) when you click submit. This can probably be disabled with logging.properties?

[image]

shroffk, 2019-08-20:
yes....sorry this is the raw filter I used for debugging

shroffk, 2019-09-23:
can we close this??

MaticPogacnik, 2019-09-23:
Yes, this can be closed. Thank you!

closed 2019-09-23 by MaticPogacnik