ControlSystemStudio/phoebus #760: xyplot background_color break zoom

mdavidsaver, 2019-08-07:
Here is a weird one.  Setting a background color for an xyplot seems to break interactive zoom.  The rubber band doesn't appear while zooming, and the scale doesn't change.

The following change alone breaks zooming.

```diff
     <name>X/Y Plot</name>
     <width>620</width>
     <height>240</height>
+    <background_color>
+      <color red="1" green="0" blue="0" alpha="0">
+      </color>
+    </background_color>
     <traces>
       <trace>
         <name>$(traces[0].y_pv)</name>
```

This is with cd749bdc6dc83e796fc8daa0606b4d994033e7a6

kasemir, 2019-08-07:
That's amazing.
But yes, I can reproduce it.
It's somehow caused by alpha=0 (transparent).
The widget becomes transparent in the sense that the mouse can no longer touch it, so move actions otherwise used for zooming no longer get to the widget.
alpha=1 is almost transparent, no visual difference, but the mouse again reaches the widget.

Not sure if this is a feature, but I'll try to figure out how to fix it.

mdavidsaver, 2019-08-12:
Confirmed.

closed 2019-08-12 by mdavidsaver